uraimo/buildSwiftOnARM #1: The libcurl patch is failing on 3.1

helje5, 2017-04-18:
Checkout seems to be sound:
```shell
helge@ZeaPro swift-corelibs-foundation (detached*)$ git status
HEAD detached at swift-3.1-RELEASE
nothing to commit, working tree clean
```

But the patch fails on it:
```shell
helge@ZeaPro swift-corelibs-foundation (detached*)$ patch -p1 < ../buildSwiftOnARM/swift-corelibs-foundation.diffs/5537ff94b5b636f451c046429cbb35e1e5b0908f.diff 
patching file CoreFoundation/Base.subproj/SwiftRuntime/CoreFoundation.h
patching file CoreFoundation/URL.subproj/CFURLSessionInterface.h
patching file build.py
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file build.py.rej
```

The patch looks like this:
```
diff --git a/build.py b/build.py
index ac4e8f7..8f1d3f8 100644
--- a/build.py
+++ b/build.py
@@ -81,7 +81,7 @@ if "XCTEST_BUILD_DIR" in Configuration.current.variables:
        ]
 
 triple = Configuration.current.target.triple
-if triple.endswith("-linux-gnu") or triple == "armv7-none-linux-androideabi":
+if triple.find("linux") != -1:
        foundation.LDFLAGS += '-lcurl '
 if triple == "armv7-none-linux-androideabi":
```

And the build.py section like this:
```
swift_cflags += [
	'-I${BUILD_DIR}/Foundation/usr/lib/swift',
	'-I${SYSROOT}/usr/include/libxml2',
	'-I${SYSROOT}/usr/include/curl'
]

if "XCTEST_BUILD_DIR" in Configuration.current.variables:
	swift_cflags += [
		'-I${XCTEST_BUILD_DIR}',
		'-L${XCTEST_BUILD_DIR}',
		'-I${SYSROOT}/usr/include/libxml2',
		'-I${SYSROOT}/usr/include/curl'
	]

triple = Configuration.current.target.triple
if triple.endswith("-linux-gnu") or triple == "armv7-none-linux-androideabi":
	foundation.LDFLAGS += '-lcurl '
```

helje5, 2017-04-18:
build.py.rej
```
***************
*** 81,87 ****
         ]
  
  triple = Configuration.current.target.triple
- if triple.endswith("-linux-gnu") or triple == "armv7-none-linux-androideabi":
         foundation.LDFLAGS += '-lcurl '
  if triple == "armv7-none-linux-androideabi":
  
--- 81,87 ----
         ]
  
  triple = Configuration.current.target.triple
+ if triple.find("linux") != -1:
         foundation.LDFLAGS += '-lcurl '
  if triple == "armv7-none-linux-androideabi":
  
```

helje5, 2017-04-18:
ffried: Because line 81 starts with a tab, then ending an array. that’s it, actually. In the patch, there are spaces, in the original there’s a tab.

Fix: invoke patch w/ `-l` aka `--ignore-whitespace`

uraimo, 2017-04-18:
Thanks @helje5, it should apply cleanly now.

commit 4cff9e84 (2017-04-18): Broken corelibs patch, fixes #1
closed 2017-04-18 by uraimo