uraimo/buildSwiftOnARM #2: Script out of box does not run successfully. Please document dependencies

scottbyrns, 2017-04-24:
`pi@raspberrypi:~/buildSwiftOnARM-master $ sudo ./build.sh 
Traceback (most recent call last):
  File "./swift/utils/build-script", line 27, in <module>
    from SwiftBuildSupport import (
  File "/home/pi/buildSwiftOnARM-master/swift/utils/SwiftBuildSupport.py", line 26, in <module>
    from swift_build_support import diagnostics  # noqa (E402)
ImportError: cannot import name diagnostics
+ Fixing up the install package for ARM
cp: cannot create directory ‘/home/pi/buildSwiftOnARM-master/install/usr/lib/swift’: No such file or directory
cp: cannot stat ‘./build/buildbot_linux/libdispatch-linux-armv7/src/swift/Dispatch.swiftdoc’: No such file or directory
cp: cannot stat ‘./build/buildbot_linux/libdispatch-linux-armv7/src/swift/Dispatch.swiftmodule’: No such file or directory
cp: cannot stat ‘./build/buildbot_linux/libdispatch-linux-armv7/src/libdispatch.la’: No such file or directory
cp: cannot stat ‘./build/buildbot_linux/libdispatch-linux-armv7/src/.libs/libdispatch.so’: No such file or directory
+ Retar installation package`

carlhung, 2017-04-24:
if you type "cat clone.sh" you can see the script will download the components and update the components.
the program is when you were downloading the components or update.
it bloody discounted all the time.
i was trying to download the components and update for whole bloody day.
and discounted more than 20 times. that's why you can't find the files when you built.
so make sure you download and update all the components successfully.
I just successfully downloaded all the files that need to build. now I can see build.sh using cmake to build swift.

scottbyrns, 2017-04-24:
That doesn't seem to be the cause for my issue. It doesn't have any problems cloning the repos.

It looks like it might not be applying the diff's that it places parallel to the repos. Investigating still.

carlhung, 2017-04-25:
I also encountered different error while compiling clang. wasted me so much time.

uraimo, 2017-04-26:
Hi @scottbyrns , yes that happens when compiling 3.1 on Raspbian.

This was fixed by @helje5 patching build-script [as show here](https://github.com/helje5/dockSwiftOnARM/blob/master/empty-ctx/rpi-ubuntu-swift-3.1.0-build.dockerfile#L41), so it's not a dependency issue (btw, it could be already fixed in master, we'll see when 4.0 lands). 

I don't have a normal patch I can give you right now since after that issue there is a blocking error (see #4) that has not been resolved yet and that I'm currently trying to debug (the repo not in a diffable state).

At the moment 3.1 compiles only on Ubuntu Mate, if you want to compile something cleanly on Raspbian you'll need to check out [the 3.0.2 tag](https://github.com/uraimo/buildSwiftOnARM/tree/3.0.2).

I'd recommend you to use Ubuntu if you can, since it's the only release that has a working SPM.

uraimo, 2017-04-29:
@scottybyrns, this is the patch for `noqa E402`: f35d8d80de3770aaa9b6da60fabd4e8bffa2b621

I've added it to the repo, but it needs to be applied manually since it's not needed on Mate.

closed 2017-04-29 by uraimo