uraimo/buildSwiftOnARM #3: Fail to compile clang on raspberry pi 3 running Lubuntu 16.04.2

carlhung, 2017-04-25:
FAILED: : && /usr/bin/clang++-3.7   -fno-stack-protector -fPIC -fvisibility-inlines-hidden -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -Wcovered-switch-default -Wnon-v
irtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -Werror=date-time -std=c++11 -fcolor-diagnostics -ffunction-sections -fdata-sections -Wdocumentation -Wimplicit-fallthrough -Wunreachable-code -Woverload
ed-virtual -DOBJC_OLD_DISPATCH_PROTOTYPES=0 -O3  -L/buildSwiftOnARM/build/buildbot_linux/swift-linux-armv7/./lib/swift/linux -target armv7-unknown-linux-gnueabihf -lpthread -latomic -ldl -fuse-ld=gold tools/swif
t-reflection-dump/CMakeFiles/swift-reflection-dump.dir/swift-reflection-dump.cpp.o  -o bin/swift-reflection-dump -L/buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/./lib -lswiftReflection /buildSwiftOnARM/
build/buildbot_linux/llvm-linux-armv7/lib/libLLVMObject.a /buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/lib/libLLVMSupport.a /buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/lib/libLLVMBitReader.a
 /buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/lib/libLLVMCore.a /buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/lib/libLLVMMCParser.a /buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/lib/l
ibLLVMMC.a /buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/lib/libLLVMSupport.a -lrt -ldl -ltinfo -lpthread -lm /buildSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/lib/libLLVMDemangle.a -Wl,-rpath,/bui
ldSwiftOnARM/build/buildbot_linux/llvm-linux-armv7/./lib && :
tools/swift-reflection-dump/CMakeFiles/swift-reflection-dump.dir/swift-reflection-dump.cpp.o:/buildSwiftOnARM/swift/tools/swift-reflection-dump/swift-reflection-dump.cpp:function swift::remote::TypeDecoder<swift
::reflection::TypeRefBuilder>::decodeMangledType(swift::Demangle::Node* const&): error: undefined reference to 'swift::Demangle::mangleNode(swift::Demangle::Node* const&)'
tools/swift-reflection-dump/CMakeFiles/swift-reflection-dump.dir/swift-reflection-dump.cpp.o:/buildSwiftOnARM/swift/tools/swift-reflection-dump/swift-reflection-dump.cpp:function swift::remote::TypeDecoder<swift
::reflection::TypeRefBuilder>::decodeMangledType(swift::Demangle::Node* const&): error: undefined reference to 'swift::Demangle::mangleNode(swift::Demangle::Node* const&)'
tools/swift-reflection-dump/CMakeFiles/swift-reflection-dump.dir/swift-reflection-dump.cpp.o:/buildSwiftOnARM/swift/tools/swift-reflection-dump/swift-reflection-dump.cpp:function swift::remote::TypeDecoder<swift
::reflection::TypeRefBuilder>::decodeMangledNominalType(swift::Demangle::Node* const&, llvm::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, swift::reflection::TypeRe
f const*&): error: undefined reference to 'swift::Demangle::mangleNode(swift::Demangle::Node* const&)'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
[774/840] Generating /buildSwiftOnAR.../swift/linux/armv7/Swift.swiftmodule
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2922:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = self
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2923:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = other
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2946:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = dividend
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2947:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = self
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2922:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = self
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2923:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = other
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2946:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = dividend
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2947:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = self
    ~~~~^~~
    _
[774/840] Compiling /buildSwiftOnARM...dlib/public/core/linux/armv7/Swift.o
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2922:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = self
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2923:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = other
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2946:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = dividend
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2947:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = self
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2922:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = self
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2923:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = other
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2946:9: warning: initialization of immutable value 'lhs' was never used; consider replacing with assignment to '_' or removing it
    let lhs = dividend
    ~~~~^~~
    _
/buildSwiftOnARM/swift/stdlib/public/core/Integers.swift.gyb:2947:9: warning: initialization of immutable value 'rhs' was never used; consider replacing with assignment to '_' or removing it
    let rhs = self
    ~~~~^~~
    _
ninja: build stopped: subcommand failed.
./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting
+ Fixing up the install package for ARM
cp: cannot create directory '/buildSwiftOnARM/install/usr/lib/swift': No such file or directory
cp: cannot stat './build/buildbot_linux/libdispatch-linux-armv7/src/swift/Dispatch.swiftdoc': No such file or directory
cp: cannot stat './build/buildbot_linux/libdispatch-linux-armv7/src/swift/Dispatch.swiftmodule': No such file or directory
cp: cannot stat './build/buildbot_linux/libdispatch-linux-armv7/src/libdispatch.la': No such file or directory
cp: cannot create regular file '/buildSwiftOnARM/install/usr/lib/swift/linux': No such file or directory
+ Retar installation package

elstntho, 2017-04-26:
Does not seem to be specific to the Raspberry: https://bugs.swift.org/browse/SR-4147

uraimo, 2017-04-26:
@carlhung thanks a lot for this issue, that's a new one (and this means that sadly, Lubuntu is not actually the same of the main flavor or Mate) and thanks @elstntho for finding something that really looks linked to this issue.

Carl can you confirm that you were compiling 3.1-RELEASE and that the swap was on?

carlhung, 2017-04-26:
how?
I followed you commands.
how can i check?

uraimo, 2017-04-26:
If you didn't change anything you are compiling `3.1-RELEASE` and that's ok.

Too check for the swap just type: `free` and verify that the swap line is something like this: `Swap:      2047996          0    2047996`, but i guess this should be ok too. I'm asking since, everyone, myself include, often forget to enable it :)

carlhung, 2017-04-27:
Swap is empty. How can I enable it? Thanks!

carlhung, 2017-04-27:
Seems ok means if I don't enable. It is still able to compile?

uraimo, 2017-04-27:
If you don't enable it, sometime along the process you will use up all the available memory and the build will crash (every time in a different place and with different errors :) ). To build it successfully you need at least 3Gb of memory between physical and swap space on the SD.

To enable it:

1. Allocate a 2Gb swap file and set the expected permissions:

           sudo fallocate -l 2G swapfile
           sudo chmod 600 swapfile

2. Format it as swap:

           sudo mkswap swapfile

3. Enable the swap:

           sudo swapon swapfile

If every step completed successfully, `free` will tell you that you have 2Gb of additional swap memory space available.

carlhung, 2017-04-27:
what about the swapfile already in the directory?

uraimo, 2017-04-27:
Oh ok, you already created it, check if it's really 2Gb in size (if not delete it and create a new one as shown above) and enable it with `sudo swapon swapfile`. You'll have to do this again if you reboot the raspberrypi (yep, you can't enable it permanently).

carlhung, 2017-04-27:
oh. as i rebooted then complied before. thanks. i will try again.

carlhung, 2017-04-27:
Still got the same error.

uraimo, 2017-04-27:
No idea of what's happening, Lubuntu should just be an ubuntu with lxde I guess, but this does not seem to be the case.

I'm compiling 3.1.1 right now on Ubuntu Mate 1604 and I should be past that point.

Just wondering, did you try if [my version](https://www.dropbox.com/s/814x8ccin363nou/swift-3.1-RPi23-1604.tgz?dl=0) for Ubuntu 16.04 worked? There is also a [3.1.1 release](https://cloud.sersoft.de/nextcloud/index.php/s/0qty8wJxlgfVCcx) compiled by @ffried.

ffried, 2017-04-27:
I had this issue as well and resolved it by specifically using clang 3.8. Not 3.7, not 3.9.
But I also heard of a case where even that didn't help.

uraimo, 2017-04-27:
Ok, I'll add a conditional at the beginning of `clone.sh` to install the correct version of the clang dependency in each case, at least in some cases it will help.

ffried, 2017-04-27:
@uraimo This has also come up in the swift-arm Slack team:
https://swift-arm.slack.com/archives/C14667C8M/p1492175049059086
It also fails on Desktop systems:
https://twitter.com/ffried/status/833985445516165121
and
https://lists.swift.org/pipermail/swift-dev/Week-of-Mon-20170424/004447.html

carlhung, 2017-04-27:
i installed the swift that you provided:
https://www.dropbox.com/s/814x8ccin363nou/swift-3.1-RPi23-1604.tgz?dl=0
got this error:
error: failed to stop process at REPL breakpoint

uraimo, 2017-04-27:
That's ok, when you launch it without parameters it simply try to start the REPL and it never worked on arm, you can try compiling single swift file with `swiftc <file>` or compile a whole project with SPM with the usual `swift build`.

carlhung, 2017-04-27:
thanks. it works. need to test.

uraimo, 2017-04-27:
Great! Feel free to report back if you find issues.

uraimo, 2018-05-13:
More than 1 year old, closing.

closed 2018-05-13 by uraimo