uraimo/buildSwiftOnARM #4: 3.1 build fails on the latest Raspbian

uraimo, 2017-04-26:
Just to document this failure (when compiling swift/ ), Bouke kindly opened an issue on jira: https://bugs.swift.org/browse/SR-4613

nesty, 2017-08-23:
On the latest Raspbian version (Stretch) released last week, Swift 3.1.1 (without swiftpm) compiles without issues (probably thanks to clang-3.8).

uraimo, 2017-08-23:
Great news @nesty, thank you for the update!
I'm going to close this, since it's not a problem anymore.

If someone wants to update their installation, the release post contains instructions, [here](https://www.raspberrypi.org/blog/raspbian-stretch/).

closed 2017-08-23 by uraimo

uraimo, 2017-08-24:
@nesty did you compile libdispatch too? And if it worked, did you build it on a fresh Raspbian Stretch image?

It looks like that Raspbian has an obsolete linux-libc-dev 3.18.5-1~exp1+rpi19+stretch that lacks the essential membarrier.h (present in newer releases).

carlhung, 2017-08-24:
i want swift package manager badly.

uraimo, 2017-08-24:
@carlhung here, Swift 3.1.1 with SPM: https://www.dropbox.com/s/v6oslfta6u773rj/swift-3.1.1-RPi23-RaspbianStretchAug17.tgz?dl=0

You'll need to do a dist-upgrade of your old Raspbian to use it (see instructions at the bottom of [this post](https://www.raspberrypi.org/blog/raspbian-stretch/)).

carlhung, 2017-08-25:
Thanks! Amazing!

nesty, 2017-08-25:
@uraimo yes I also did compile libdispatch successfully after adding the missing membarrier.h file and editing unistd.h to add the missing membarrier syscall.

Thank you for the update to compile swiftpm successfully on Raspbian Stretch 👍.

carlhung, 2017-08-25:
Will you release for pi zero? As i just got a pi zero.

uraimo, 2017-08-26:
Since it takes nearly 2 weeks, I think I'll wait for High Sierra release (19th sept?) to build directly 3.2 when they create a tag on the swift-3.1-branch.