uraimo/buildSwiftOnARM #5: Swift 4.0 Status

uraimo, 2017-08-23:
I've made an attempt (17th july, ref 5c06e85) at compiling 4.0 from swift-4.0-branch on Ubuntu Mate and everything compiled except for SPM:

https://www.dropbox.com/s/ykirsy3ylbaxxmd/swift-4.0-Rpi23-1604-snapshot-5c06e85.tgz?dl=0

After the move to Stretch, the result should be the same on Raspbian.

uraimo, 2017-09-06:
See: [SR-5845](https://bugs.swift.org/browse/SR-5845), [SR-5846](https://bugs.swift.org/browse/SR-5846).

Ybrin, 2017-11-29:
Do you have any update or workaround for this issue? I just recently tried building swift-4.0.2-release and I got the exact same error.

uraimo, 2017-11-30:
Sadly no, even just debugging the first issue is proving quite time-consuming due to hardware constraints on the Pi (after *multiple* attempts I've got a build with few more assertions but they don't give new info, not even on the weakref issue). 
I have some updates from a few weeks ago irt new tests failing on 4.0.2-release but no fixes or workarounds.
I'd really like to see this working too but right now I just have the time to keep an eye on 4.1 for additional issues that could be introduced with the next release on top of these and not much else.

Dev1an, 2017-12-10:
Did anyone look at what @zbowling is doing? He is working in the [google/swift](https://github.com/google/swift/tree/multi_arch) fork and [it seems](https://github.com/apple/swift/pull/12955#issuecomment-344782406) that he is planning to add ARM support for swift and is dedicated to get his changes pulled back into the main (apple/swift) repository.

The changes he is making are targeting swift for Fuchsia, but will this also help us to get swift-4 on Ubuntu?

zbowling, 2017-12-11:
I'm getting side by side arch libraries working in Swift so single toolchain can cross compile for a number of archs (it wasn't possible previously because the cmake system was built to assume everything worked like Darwin which has lipo). For Fuchsia I only care about aarch64 and x86_64 working.

uraimo, 2017-12-29:
Late reply, sorry. Yes, I didn't see anything there that could help with the issues on 32bit arm, but I'll 
 verify that the build will still start on armv6/7 when it gets merged.

compnerd, 2018-01-23:
@uraimo, perhaps setting up a dashboard on Jira would be helpful.  At least that way we could track what is known to be an issue with Linux ARM.  I have been working on the Linux AArch64 support on and off for a while, and it is nearly complete (it is missing support for variadics, which I have a partial patch that needs to be completed).

uraimo, 2018-01-23:
Yes, it makes a lot of sense, it's available [here](https://bugs.swift.org/secure/Dashboard.jspa?selectPageId=10900) and it should be accessible to everyone.
I will close those that are not reproducible on current versions of either Ubuntu or Raspbian and update a few of them in the next few days.

Dev1an, 2018-05-04:
It looks like swift 4 is back on arm: https://medium.com/@futurejones/1e20ba5a586e

uraimo, 2018-05-04:
Yup! @chnmrc got the last piece of the puzzle working (spm), testing in progress.

A new issue popped up on master but shouldn't be hard to fix.

patchthecode, 2018-05-04:
@Dev1an hmm. isnt that article about Swift NIO? unless i misunderstand.

helje5, 2018-05-04:
NIO is the headline of that article, but the key thing is that @chnmrc got 4.1 working on Raspi 32bit (and @futurejones on Raspi 64bit). 4.x is a precondition for NIO.

uraimo, 2018-05-13:
Now we have at least 4.1.1 building correctly on Ubuntu (Raspbian v6/v7 running right now, someone else reported success on linux/aarch64), so I'm closing this!

closed 2018-05-13 by uraimo