uraimo/buildSwiftOnARM #7: Building Swift 4.1 fails

adtrevor, 2018-05-12:
I followed the latest instructions to build Swift 4.1 but had this error close to the end :
swift-build-stage1: malloc.c:2394: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
--- bootstrap: error: build failed with exit status -6
./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting

adtrevor, 2018-05-12:
This is on a Raspberry with Linux Ubuntu Mate 16.04.4

uraimo, 2018-05-12:
I've seen that error before but can't recall what was the cause. The only thing that I've changed from the instructions is that I'm using 3Gb of swap now, but since you've already reached swiftpm that shouldn't be a problem.
Did you start from scratch with the repos aligned to swift-4.1-RELEASE (that's what checkoutRelease did before the last commit that moved to 4.1.1)?
If you already had something in `build/` , I'd try again removing `build/buildbot_linux/swiftpm-linux-armv7` to do a clean build of swiftpm.

adtrevor, 2018-05-12:
No I downloaded and did checkoutRelease yesterday before your commit of today, I will try removing the swiftpm builddir.

I have about 16 GB of swap on a usb stick.

Thank you !

Trevör

adtrevor, 2018-05-12:
Removed the spm builddir and it didn’t work, there is still the same issue unfortunately, I will probably clean pull your last changes and do a checkOutRelease again.

Thank you !

Trevör

adtrevor, 2018-05-13:
I pulled your changes, ran checkoutRelease.sh, didn't clean and built again and it worked ! Now I have Swift 4.1.1 ! Thank you ! :)

closed 2018-05-13 by adtrevor

uraimo, 2018-05-13:
Great! I guess it's just one of those transient failures that happen from time to time and rebuilding fixes it.

adtrevor, 2018-05-13:
Yes ! Now I am trying some Swift packages such as https://github.com/IBM-Swift/Swift-SMTP, unfortunately I got a segfault, do you know if lldb works properly on Linux ARM ? I tried to debug my issue but the breakpoints don't seem to work :/

uraimo, 2018-05-13:
Have you tried with gdb?

adtrevor, 2018-05-13:
Nope, but does gdb support Swift ?

uraimo, 2018-05-13:
No afaik, but it could be worth a try if lldb does not seem to work, it could give some hints about what’s happening.

adtrevor, 2018-05-13:
I will probably try that, in the meantime I have really unstable Swift, for instance even this:
```
import Foundation
if let str = try? String(contentsOfFile: "/path/to/some/text/file")  {
    print(str)
}
```

Fails with illegal instruction

adtrevor, 2018-05-13:
Do you think that it might have to do with the fact that I didn't clean before building again ? I thought it would detect what had changed and build only what's needed, do you think I could be wrong ? Also I don't remember seeing it build llvm when I went from 4.1 to 4.1.1...

adtrevor, 2018-05-13:
Apparently llvm and cmark build folders were untouched after the upgrade to Swift 4.1.1

uraimo, 2018-05-13:
Yup, it's possible that nothing changed between the two tags.
For the error, it's not you,  i see the same thing with my last build.

uraimo, 2018-05-13:
If you want you can join the swift-arm slack: https://launchpass.com/swift-arm