uraimo/buildSwiftOnARM #11: Strange behavior with Swift 4.1.1 in Raspbian

jdrobert, 2018-05-22:
Hello,

I ran the scripts on a Raspberry Pi 3 running Raspbian lite Stretch. Everything went pretty smoothly except for an error with Swig because I had 3.0.10 installed. I built and installed Swig 3.0.12 but still had to manually update the path in lldb/scripts/CMakeLists.txt. After that everything completed successfully. [I also attached my`build.sh` output.](https://github.com/uraimo/buildSwiftOnARM/files/2028958/nohup.log)

Now on to my actual problem. I created a project using Swift Package Manager (`swift package init --type executable`). In main.swift, if I try to create a URL object everything works as expected. 

```Swift
import Foundation

let url = URL(string:"http://www.google.com")
print(url)

```

This prints out `Optional(http://www.google.com)`

However if I try to do the same thing from a class (either within main.swift or a new file), I get `Illegal instruction`

```Swift
import Foundation

class MakeAURL {
        func makeURL() {
                let url = URL(string: "http://www.google.com")
                print(url)
        }
}

let url = URL(string:"http://www.google.com")
print(url)

print("This does show up")

let makeAURL = MakeAURL()
makeAURL.makeURL()

print("This doesn't show up")
```

This prints out:
```
Optional(http://www.google.com)
This does show up
Optional(http://www.google.com)
Illegal instruction
```

Was there a patch that I should have applied manually? Any help would be appreciated. 

Thank you,
--Jeremy

uraimo, 2018-05-23:
Hi Jeremy, thank you for a minimal sample I could test.
Did your build include [this workaround](https://github.com/uraimo/buildSwiftOnARM/blob/master/swift-corelibs-foundation.diffs/string01.diff) added 3 days ago? I suppose the answer is affermative.

Can you try this [patch](https://github.com/uraimo/buildSwiftOnARM/files/2030078/urlfix.diff.zip) and let me know if something changed? (`patch -p1 < urlfix.diff`)

jdrobert, 2018-05-23:
I'm pretty sure that workaround was not included when I built. I can't find `string01.diff` I'll apply the url patch and rebuild. It may take me a day or two to get back with you. Thanks for your help.

uraimo, 2018-05-23:
I've checked, string01.diff should be enough to make it work.

jdrobert, 2018-05-24:
That seems to have worked. Thanks for your help. I'll keep going and let you know if I run into any more problems.

closed 2018-05-24 by jdrobert