uraimo/buildSwiftOnARM #15: [4.1.1] URLSession.dataTask causes "Segmentation fault"

jdrobert, 2018-05-25:
Thanks for helping me with my previous issue. This may have the same underlying issue as #12 but I figured it wouldn't hurt to document it. 

This code worked in the armv6 3.1 binaries I downloaded from your blog but it is crashing in 4.1.1 on armv7. It also looks like it was a known issue in 3.0 - https://bugs.swift.org/browse/SR-3365

```Swift 
import Foundation
import Dispatch

var urlRequest = URLRequest(url: URL(string:"https://httpbin.org/get")!)
urlRequest.httpMethod = "GET"
let config = URLSessionConfiguration.default
let session = URLSession(configuration: config)

let semaphore = DispatchSemaphore(value: 0)
let task = session.dataTask(with: urlRequest, completionHandler: { (data, response, error) in    
    print(String(data:data!, encoding:.utf8))
    semaphore.signal()
})
task.resume()

_ = semaphore.wait(timeout: .distantFuture)
```

On my Pi3 this prints out `Segmentation fault` but on my Mac I get the correct response. 

Thank you,
Jeremy

uraimo, 2018-05-25:
Hi! Thanks for the sample, I'll try it out to see if it's related to the other one or not.

NobodyNada, 2018-06-09:
Both this and #12 were caused by Foundation assuming an incorrect memory offset when bridging a constant CFString, [here](https://github.com/apple/swift-corelibs-foundation/blob/master/Foundation/NSCFString.swift#L61).  This appears to have been [fixed](https://github.com/apple/swift-corelibs-foundation/pull/1518/files#diff-07b7205addeb35f2175a7b1de35289a0R71) in [this PR](https://github.com/apple/swift-corelibs-foundation/pull/1518), which it looks like did not quite make it into Swift 4.1.

uraimo, 2018-06-10:
Thanks @NobodyNada, I've checked those diffs, especially the changes to `_lenOffset` that I missed the first time I saw them, and I was thinking to extract some of them to build a minimal patch.
Sadly, those modifications have also not been merged in swift-4.2-branch.

uraimo, 2018-09-13:
Fixed, even though we should verify if `dataTask` actually works (and if not, if the functionality is actually implemented)

closed 2018-09-13 by uraimo

jdrobert, 2018-09-13:
Thank you. I’ll probably have time this weekend to try again.