uraimo/buildSwiftOnARM #16: #if os(Linux) not working?

Schaefers, 2018-06-03:
Hi, I'm trying to compile https://github.com/IBM-Swift/SwiftyJSON,

but every `#if os(Linux)` seems to be false, so for lines like 
```swift
#if os(Linux)
    return lhs.rawArray._bridgeToObjectiveC() == rhs.rawArray._bridgeToObjectiveC()
#else
    return lhs.rawArray as NSArray == rhs.rawArray as NSArray
#endif
```
https://github.com/IBM-Swift/SwiftyJSON/blob/master/Sources/SwiftyJSON/SwiftyJSON.swift#L1463

I get 
```
error: cannot convert value of type '[Any]' to type 'NSArray' in coercion
        return lhs.rawArray as NSArray == rhs.rawArray as NSArray
               ~~~~^~~~~~~~
```

as error.

uraimo, 2018-06-08:
Hi, at least on Mate that #if is working correctly, can you do a quick check and verify that this works?

```
#if os(Linux)
        print("Linux!")
#else
        print("Not Linux?")
#endif
```

Schaefers, 2018-06-08:
Strange, this is working...

uraimo, 2018-06-10:
No idea (misaligned macros? improbable), you could try with SwiftyJSON original repository instead of this fork from IBM.

keybuk, 2018-06-11:
Note that #if does not prevent the compiler from parsing and validating the #else clause, unlike compilers that use the C preprocessor - so even on Linux, swiftc will check the other side, which is what it's doing here.

uraimo, 2018-06-11:
Thanks for pointing that out, btw @Schaefers, when do you get that error? Compiling IBM's swiftyjson in isolation with 4.1.2 works for me.

Schaefers, 2018-06-11:
I'm using it in an Swift Package Manager project, with the `Package.swift` file underneath and compiling using `swift build`.

```
// swift-tools-version:4.0
// The swift-tools-version declares the minimum version of Swift required to build this package.

import PackageDescription

let package = Package(
    name: "removed",
    dependencies: [
        // Dependencies declare other packages that this package depends on.
        // .package(url: /* package url */, from: "1.0.0"),
        .package(url: "https://github.com/uraimo/SwiftyGPIO.git", .branch("master")),
        .package(url: "https://github.com/IBM-Swift/SwiftyRequest.git", .upToNextMajor(from: "1.0.0")),
        .package(url: "https://github.com/SwiftyJSON/SwiftyJSON.git", .branch("master"))
    ],
    targets: [
        // Targets are the basic building blocks of a package. A target can define a module or a test suite.
        // Targets can depend on other targets in this package, and on products in packages which this package depends on.
        .target(
            name: "removed",
            dependencies: ["SwiftyGPIO", "SwiftyRequest", "SwiftyJSON"]),
        ]
)
```

Schaefers, 2018-06-18:
Compiling *IBM-Swift/SwiftyJSON* from source works for me as well. So is there a problem with the package manager or my `Package.swift` file?

Kaiede, 2018-09-12:
Are you still trying to use SwiftyJSON/SwiftyJSON? I've never gotten that to build on Linux for me, only the IBM-Swift fork. 

I'm currently able to build https://github/Kaiede/RPiLight which uses IBM-Swift/SwiftyJSON on both Swift 3.1.1 and Swift 4.1.2 without problem.

Schaefers, 2018-09-12:
I have to double check, if I really messed up my `Package.swift` file. As stated in the first post, my intention was to use the IBM fork..

Kaiede, 2018-09-12:
About the only difference is that I resolve using: .package(url: "https://github.com/IBM-Swift/SwiftyJSON.git", from: "17.0.2")

But the difference shouldn't really matter. Resolving to master should should be fine in this case. 

The catch is that there's a good chunk of work in the official repo that IBM never merged. But IBM never merged the Linux-specific stuff back either. *sigh*

closed 2019-02-13 by uraimo