uraimo/buildSwiftOnARM #18: ./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting

Hackberger, 2018-06-22:
Hello!

I am running into slight issue at presumably near the end of the build process. I appreciate any help and directions given.

Many thanks in advance!

proc/cpuinfo:
```
processor	: 0 through 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
```

This is how it end:

```
[812/868] Performing configure step for 'libdispatch'
-- The C compiler identification is Clang 5.0.0
-- The CXX compiler identification is Clang 5.0.0
-- Check for working C compiler: /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/bin/clang
-- Check for working C compiler: /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/bin/clang++
-- Check for working CXX compiler: /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Looking for __GNU_LIBRARY__
-- Looking for __GNU_LIBRARY__ - found
-- Performing Test __BUILTIN_TRAP
-- Performing Test __BUILTIN_TRAP - Success
-- Found LibRT: /usr/lib/arm-linux-gnueabihf/librt.so  
-- Looking for _pthread_workqueue_init
-- Looking for _pthread_workqueue_init - not found
-- Looking for getprogname
-- Looking for getprogname - not found
-- Looking for mach_absolute_time
-- Looking for mach_absolute_time - not found
-- Looking for mach_approximate_time
-- Looking for mach_approximate_time - not found
-- Looking for mach_port_construct
-- Looking for mach_port_construct - not found
-- Looking for malloc_create_zone
-- Looking for malloc_create_zone - not found
-- Looking for pthread_key_init_np
-- Looking for pthread_key_init_np - not found
-- Looking for pthread_main_np
-- Looking for pthread_main_np - not found
-- Looking for pthread_workqueue_setdispatch_np
-- Looking for pthread_workqueue_setdispatch_np - not found
[... 588 lines cut ...]
FAILED: tests/dispatch_timer_bit31 
: && /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/bin/clang    -lrt tests/CMakeFiles/dispatch_timer_bit31.dir/dispatch_timer_bit31.c.o  -o tests/dispatch_timer_bit31  -Wl,-rpath,/home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/src:/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7 src/libdispatch.so libBlocksRuntime.a -L/home/linuxbrew/.linuxbrew/Cellar/libbsd/0.8.3/lib -lbsd /home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so /home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftSwiftOnoneSupport.so tests/libbsdtests.a -pthread && :
/usr/bin/ld: warning: libicuuc.so.61, needed by /home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libicui18n.so.61, needed by /home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so, not found (try using -rpath or -rpath-link)
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `uiter_setString_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ubrk_preceding_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ucol_setAttribute_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `u_strToLower_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ucol_openElements_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ucol_strcollIter_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `uiter_setUTF8_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `u_strToUpper_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ubrk_following_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ubrk_setText_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ubrk_open_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ucol_closeElements_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ucol_next_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ucol_strcoll_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ubrk_close_61'
/home/pi/swift/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/armv7/libswiftCore.so: undefined reference to `ucol_open_61'
clang-5.0: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting
```

uraimo, 2018-06-22:
Hi! Can you verify if libicu is installed with:

        sudo apt-get install libicu-dev icu-devtools

Hackberger, 2018-06-22:
It appears it is. Output is in German and it says that 

- icu-devtools is already the newest version
- libicu-dev is already the newest version

```
:~ $ sudo apt-get install libicu-dev icu-devtools
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
icu-devtools ist schon die neueste Version (57.1-6+deb9u2).
libicu-dev ist schon die neueste Version (57.1-6+deb9u2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 2 nicht aktualisiert.
```

uraimo, 2018-06-22:
Same versions on my Pi3, `libicu` should also have been installed automatically since it's a dependence of icu-dev.
Weird that it tries to link to .61, can you confirm that this is a clean build made with 4.1.2 without any modifications?

Hackberger, 2018-06-22:
I tried it before and it failed with due to a full disk.

Then I ran in given order

1. clean.sh
2. checkoutRelease.sh
3. build.sh

Is there anything I can try?

uraimo, 2018-06-22:
Just to be sure, you are using Raspbian right?

What does `apt list --installed | grep icu" says? If for example, you have icu.61 installed you could try removing it.

Hackberger, 2018-06-22:
Yes, Raspbian.

```
: $ apt list --installed | grep icu

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

icu-devtools/stable,now 57.1-6+deb9u2 armhf  [installiert]
libbasicusageenvironment1/stable,now 2016.11.28-1 armhf  [Installiert,automatisch]
libharfbuzz-icu0/stable,now 1.4.2-1 armhf  [Installiert,automatisch]
libicu-dev/stable,now 57.1-6+deb9u2 armhf  [installiert]
libicu57/stable,now 57.1-6+deb9u2 armhf  [Installiert,automatisch]
```

Doesn't say anything about icu.61.

uraimo, 2018-06-22:
Oh... you are using linuxbrew, so I guess that .61 can be found but cannot be properly linked.
Run this:

        brew info icu4c

If this is similar to the macOS brew, it should display the path that needs to be added to LD_FLAGS, or you can just run `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:<your icu path here>` right before starting a build. This should fix it.

Hackberger, 2018-06-22:
So I did this.

```
: $ brew info icu4c
/home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/brew.sh: line 4: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
icu4c: stable 61.1, HEAD
C/C++ and Java libraries for Unicode and globalization
http://site.icu-project.org/
/home/linuxbrew/.linuxbrew/Cellar/icu4c/61.1 (249 files, 68.4MB) *
  Built from source on 2018-06-12 at 14:31:46
From: https://github.com/Linuxbrew/homebrew-core/blob/master/Formula/icu4c.rb
==> Options
--HEAD
	Install HEAD version

```
And this

`: $ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/linuxbrew/.linuxbrew/Cellar/icu4c/61.1`

And it is still not working.

```
: $ ./build.sh 
+ mkdir -p /home/pi/swift/build/buildbot_linux
+ env HOST_VARIABLE_linux_armv7__SWIFT_BENCHMARK_TARGETS= HOST_VARIABLE_linux_armv7__SWIFT_RUN_BENCHMARK_TARGETS= HOST_VARIABLE_linux_armv7__SWIFT_SDKS=LINUX HOST_VARIABLE_linux_armv7__SWIFT_STDLIB_TARGETS=swift-test-stdlib-linux-armv7 HOST_VARIABLE_linux_armv7__SWIFT_TEST_TARGETS= /home/pi/swift/swift/utils/build-script-impl --workspace /home/pi/swift --build-dir /home/pi/swift/build/buildbot_linux --install-prefix /usr --host-target linux-armv7 --stdlib-deployment-targets linux-armv7 --host-cc /usr/bin/clang --host-cxx /usr/bin/clang++ --darwin-xcrun-toolchain default --darwin-deployment-version-osx=10.9 --darwin-deployment-version-ios=7.0 --darwin-deployment-version-tvos=9.0 --darwin-deployment-version-watchos=2.0 --cmake /usr/bin/cmake --cmark-build-type Release --llvm-build-type Release --swift-build-type Release --swift-stdlib-build-type Release --lldb-build-type Release --foundation-build-type Release --libdispatch-build-type Release --libicu-build-type Release --xctest-build-type Release --swiftpm-build-type Release --swift-enable-assertions true --swift-stdlib-enable-assertions true --swift-analyze-code-coverage false --cmake-generator Ninja --build-jobs 4 '--common-cmake-options=-G Ninja -DCMAKE_C_COMPILER:PATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:PATH=/usr/bin/clang++ -DLLVM_VERSION_MAJOR:STRING=5 -DLLVM_VERSION_MINOR:STRING=0 -DLLVM_VERSION_PATCH:STRING=0 -DCLANG_VERSION_MAJOR:STRING=5 -DCLANG_VERSION_MINOR:STRING=0 -DCLANG_VERSION_PATCH:STRING=0 -DCMAKE_MAKE_PROGRAM=/usr/bin/ninja' --build-args=-j4 --cmark-cmake-options= '--llvm-cmake-options=-DLLVM_ENABLE_ASSERTIONS=TRUE -DLLVM_TARGETS_TO_BUILD=X86;ARM;AArch64;PowerPC;SystemZ;Mips' '--swift-cmake-options=-DSWIFT_STDLIB_ENABLE_SIL_OWNERSHIP=FALSE -DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -DSWIFT_FORCE_OPTIMIZED_TYPECHECKER=FALSE' --lldb-cmake-options= --llbuild-cmake-options= --libdispatch-cmake-options= --foundation-cmake-options= --xctest-cmake-options= --swiftpm-cmake-options= --build-stdlib-deployment-targets all --ninja-bin=/usr/bin/ninja --skip-build-libicu --skip-build-playgroundsupport --build-swift-dynamic-stdlib --build-swift-static-stdlib --build-swift-dynamic-sdk-overlay --build-swift-static-sdk-overlay --skip-build-ios-device --skip-build-ios-simulator --skip-build-tvos-device --skip-build-tvos-simulator --skip-build-watchos-device --skip-build-watchos-simulator --skip-build-android --skip-test-swift --skip-test-cmark --skip-test-lldb --skip-test-llbuild --skip-test-swiftpm --skip-test-xctest --skip-test-foundation --skip-test-libdispatch --skip-test-libicu --skip-test-playgroundsupport --skip-test-linux --skip-test-freebsd --skip-test-cygwin --skip-test-osx --skip-test-ios-host --skip-test-ios-simulator --skip-test-tvos-host --skip-test-tvos-simulator --skip-test-watchos-host --skip-test-watchos-simulator --skip-test-android-host --skip-test-benchmarks --skip-test-optimized --skip-test-optimize-for-size --android-deploy-device-path /data/local/tmp --install-libdispatch --install-foundation --install-swift --install-lldb --install-llbuild --install-xctest --install-swiftpm '--swift-install-components=autolink-driver;compiler;clang-builtin-headers;stdlib;swift-remote-mirror;sdk-overlay;dev' --install-destdir=/home/pi/swift/install --installable-package=/home/pi/swift/swift-4.1.2.tgz --llvm-lit-args=-sv
Building the standard library for: swift-test-stdlib-linux-armv7
cmark: using gold linker
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/cmark-linux-armv7 -- -j4 all
ninja: no work to do.
llvm: using gold linker
symlinking the system headers (/usr/include/c++) into the local clang build directory (/home/pi/swift/build/buildbot_linux/llvm-linux-armv7/include).
+ ln -s -f /usr/include/c++ /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/include
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/llvm-linux-armv7 -- -j4 all
ninja: no work to do.
swift: using gold linker
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/swift-linux-armv7 -- -j4 all swift-test-stdlib-linux-armv7
[1/12] Performing build step for 'libdispatch'
FAILED: tools/SourceKit/libdispatch-prefix/src/libdispatch-stamp/libdispatch-build /home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/src/libdispatch.so /home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/libBlocksRuntime.a 
cd /home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7 && /usr/bin/cmake --build .
[1/36] Linking C executable tests/dispatch_api
FAILED: tests/dispatch_api 
: && /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/bin/clang    -lrt tests/CMakeFiles/dispatch_api.dir/dispatch_api.c.o  -o tests/dispatch_api  -Wl,-rpath,/home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/src:/home/pi/swift/build/ [... cut ...]
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 0 changed, 0 removed
looking for now-outdated files... none found
no targets are out of date.
build succeeded.
ninja: build stopped: subcommand failed.
./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting
```

It is still not working. 

I used `brew` to install the NextCloud client. But I do not require it any more. So if we need to break anything here in order to get swift working, alright with me.

uraimo, 2018-06-22:
On a mac, that path is `/usr/local/opt/icu4c/lib` and you'll see it's the right one because in that directory you have files like `libicuuc.a`,etc... , my guess is that the Cellar path is wrong. Maybe add a `/lib` at the end like clang is doing for `-L/home/linuxbrew/.linuxbrew/Cellar/libbsd/0.8.3/lib`?

If you can uninstall it, `brew uninstall icu4c` *should* suffice.
This error:

    /usr/bin/ld: warning: libicuuc.so.61, needed by /home/pi/swift/build/build.......

Should disappear and ld should be able to link those libdispatch binaries.

Hackberger, 2018-06-22:
Great! I jumped this hurdle. Many thanks so far!

I removed icu4c.

` $ brew uninstall icu4c
`
But then, I figured, I would have had to re-compile and link everything again. The `build.sh` also ended immediately with an error.

So I re-installed icu4c.

` $ brew install icu4c`

And I added the `/lib` at the end of the path. This did the trick!

It did a lot of linking and installing but, alas, later on some issue with python scripting came across. Any ideas?

I really appreciate your patience and assistance!

```
 $ ./build.sh 
+ mkdir -p /home/pi/swift/build/buildbot_linux
+ env HOST_VARIABLE_linux_armv7__SWIFT_BENCHMARK_TARGETS= HOST_VARIABLE_linux_armv7__SWIFT_RUN_BENCHMARK_TARGETS= HOST_VARIABLE_linux_armv7__SWIFT_SDKS=LINUX HOST_VARIABLE_linux_armv7__SWIFT_STDLIB_TARGETS=swift-test-stdlib-linux-armv7 HOST_VARIABLE_linux_armv7__SWIFT_TEST_TARGETS= /home/pi/swift/swift/utils/build-script-impl --workspace /home/pi/swift --build-dir /home/pi/swift/build/buildbot_linux --install-prefix /usr --host-target linux-armv7 --stdlib-deployment-targets linux-armv7 --host-cc /usr/bin/clang --host-cxx /usr/bin/clang++ --darwin-xcrun-toolchain default --darwin-deployment-version-osx=10.9 --darwin-deployment-version-ios=7.0 --darwin-deployment-version-tvos=9.0 --darwin-deployment-version-watchos=2.0 --cmake /usr/bin/cmake --cmark-build-type Release --llvm-build-type Release --swift-build-type Release --swift-stdlib-build-type Release --lldb-build-type Release --foundation-build-type Release --libdispatch-build-type Release --libicu-build-type Release --xctest-build-type Release --swiftpm-build-type Release --swift-enable-assertions true --swift-stdlib-enable-assertions true --swift-analyze-code-coverage false --cmake-generator Ninja --build-jobs 4 '--common-cmake-options=-G Ninja -DCMAKE_C_COMPILER:PATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:PATH=/usr/bin/clang++ -DLLVM_VERSION_MAJOR:STRING=5 -DLLVM_VERSION_MINOR:STRING=0 -DLLVM_VERSION_PATCH:STRING=0 -DCLANG_VERSION_MAJOR:STRING=5 -DCLANG_VERSION_MINOR:STRING=0 -DCLANG_VERSION_PATCH:STRING=0 -DCMAKE_MAKE_PROGRAM=/usr/bin/ninja' --build-args=-j4 --cmark-cmake-options= '--llvm-cmake-options=-DLLVM_ENABLE_ASSERTIONS=TRUE -DLLVM_TARGETS_TO_BUILD=X86;ARM;AArch64;PowerPC;SystemZ;Mips' '--swift-cmake-options=-DSWIFT_STDLIB_ENABLE_SIL_OWNERSHIP=FALSE -DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -DSWIFT_FORCE_OPTIMIZED_TYPECHECKER=FALSE' --lldb-cmake-options= --llbuild-cmake-options= --libdispatch-cmake-options= --foundation-cmake-options= --xctest-cmake-options= --swiftpm-cmake-options= --build-stdlib-deployment-targets all --ninja-bin=/usr/bin/ninja --skip-build-libicu --skip-build-playgroundsupport --build-swift-dynamic-stdlib --build-swift-static-stdlib --build-swift-dynamic-sdk-overlay --build-swift-static-sdk-overlay --skip-build-ios-device --skip-build-ios-simulator --skip-build-tvos-device --skip-build-tvos-simulator --skip-build-watchos-device --skip-build-watchos-simulator --skip-build-android --skip-test-swift --skip-test-cmark --skip-test-lldb --skip-test-llbuild --skip-test-swiftpm --skip-test-xctest --skip-test-foundation --skip-test-libdispatch --skip-test-libicu --skip-test-playgroundsupport --skip-test-linux --skip-test-freebsd --skip-test-cygwin --skip-test-osx --skip-test-ios-host --skip-test-ios-simulator --skip-test-tvos-host --skip-test-tvos-simulator --skip-test-watchos-host --skip-test-watchos-simulator --skip-test-android-host --skip-test-benchmarks --skip-test-optimized --skip-test-optimize-for-size --android-deploy-device-path /data/local/tmp --install-libdispatch --install-foundation --install-swift --install-lldb --install-llbuild --install-xctest --install-swiftpm '--swift-install-components=autolink-driver;compiler;clang-builtin-headers;stdlib;swift-remote-mirror;sdk-overlay;dev' --install-destdir=/home/pi/swift/install --installable-package=/home/pi/swift/swift-4.1.2.tgz --llvm-lit-args=-sv
Building the standard library for: swift-test-stdlib-linux-armv7
cmark: using gold linker
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/cmark-linux-armv7 -- -j4 all
ninja: no work to do.
llvm: using gold linker
symlinking the system headers (/usr/include/c++) into the local clang build directory (/home/pi/swift/build/buildbot_linux/llvm-linux-armv7/include).
+ ln -s -f /usr/include/c++ /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/include
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/llvm-linux-armv7 -- -j4 all
ninja: no work to do.
swift: using gold linker
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/swift-linux-armv7 -- -j4 all swift-test-stdlib-linux-armv7
[1/12] Building HTML documentation
Running Sphinx v1.4.9
loading pickled environment... done
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 0 changed, 0 removed
looking for now-outdated files... none found
no targets are out of date.
build succeeded.
[2/12] Performing build step for 'libdispatch'
[1/36] Linking C executable tests/dispatch_debug
[2/36] Linking C executable tests/dispatch_timer_bit31
[3/36] Linking C executable tests/dispatch_api
[4/36] Linking C executable tests/bsdtestharness
[5/36] Linking CXX executable tests/dispatch_plusplus
[6/36] Linking C executable tests/dispatch_overcommit
[7/36] Linking C executable tests/dispatch_context_for_key
[8/36] Building C object tests/CMakeFiles/dispatch_sema.dir/dispatch_sema.c.o
[... cut ...]
/usr/bin/swig3.0 -c++ -shadow -python -threads -I/home/pi/swift/lldb/include -I. -D__STDC_LIMIT_MACROS -D__STDC_CONSTANT_MACROS -outdir /home/pi/swift/build/buildbot_linux/lldb-linux-armv7/scripts -o /home/pi/swift/build/buildbot_linux/lldb-linux-armv7/scripts/LLDBWrapPython.cpp /home/pi/swift/lldb/scripts/lldb.swig
[7/906] Building CXX object source/Core/CMakeFiles/lldbCore.dir/DumpDataExtractor.cpp.o
ninja: build stopped: subcommand failed.
./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting
```

uraimo, 2018-06-22:
Check out the bottom of this post, you need to update swig manually ;)
https://www.uraimo.com/2018/06/13/A-big-update-on-Swift-4-1-2-for-raspberry-pi-zero-1-2-3/

Hackberger, 2018-06-22:
It has continued to build! Thanks for the hint.

Regarding your blog post: it appears that the wget link should be

```
wget http://ftp.de.debian.org/debian/pool/main/s/swig/swig3.0_3.0.12-1.1_armhf.deb
wget http://ftp.de.debian.org/debian/pool/main/s/swig/swig_3.0.12-1.1_armhf.deb

```

You seem to have missed a `.1` inside the file names or the versions have changed meanwhile.

Otherwise it works fine so far! I will close the issue once it finishes. Fingers crossed.

uraimo, 2018-06-22:
Oh, thanks! Looks like they have updated the packages and removed the old ones :/
I'm closing this, but feel free to comment anyway if it stops somewhere else (it shouldn't).

closed 2018-06-22 by uraimo

Hackberger, 2018-06-25:
In fact it stopped again.

Maybe you can have another look into it! It fails on finding `libcurl` (look at the end of this post).

```
 $ ./build.sh 
+ mkdir -p /home/pi/swift/build/buildbot_linux
+ env HOST_VARIABLE_linux_armv7__SWIFT_BENCHMARK_TARGETS= HOST_VARIABLE_linux_armv7__SWIFT_RUN_BENCHMARK_TARGETS= HOST_VARIABLE_linux_armv7__SWIFT_SDKS=LINUX HOST_VARIABLE_linux_armv7__SWIFT_STDLIB_TARGETS=swift-test-stdlib-linux-armv7 HOST_VARIABLE_linux_armv7__SWIFT_TEST_TARGETS= /home/pi/swift/swift/utils/build-script-impl --workspace /home/pi/swift --build-dir /home/pi/swift/build/buildbot_linux --install-prefix /usr --host-target linux-armv7 --stdlib-deployment-targets linux-armv7 --host-cc /usr/bin/clang --host-cxx /usr/bin/clang++ --darwin-xcrun-toolchain default --darwin-deployment-version-osx=10.9 --darwin-deployment-version-ios=7.0 --darwin-deployment-version-tvos=9.0 --darwin-deployment-version-watchos=2.0 --cmake /usr/bin/cmake --cmark-build-type Release --llvm-build-type Release --swift-build-type Release --swift-stdlib-build-type Release --lldb-build-type Release --foundation-build-type Release --libdispatch-build-type Release --libicu-build-type Release --xctest-build-type Release --swiftpm-build-type Release --swift-enable-assertions true --swift-stdlib-enable-assertions true --swift-analyze-code-coverage false --cmake-generator Ninja --build-jobs 4 '--common-cmake-options=-G Ninja -DCMAKE_C_COMPILER:PATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:PATH=/usr/bin/clang++ -DLLVM_VERSION_MAJOR:STRING=5 -DLLVM_VERSION_MINOR:STRING=0 -DLLVM_VERSION_PATCH:STRING=0 -DCLANG_VERSION_MAJOR:STRING=5 -DCLANG_VERSION_MINOR:STRING=0 -DCLANG_VERSION_PATCH:STRING=0 -DCMAKE_MAKE_PROGRAM=/usr/bin/ninja' --build-args=-j4 --cmark-cmake-options= '--llvm-cmake-options=-DLLVM_ENABLE_ASSERTIONS=TRUE -DLLVM_TARGETS_TO_BUILD=X86;ARM;AArch64;PowerPC;SystemZ;Mips' '--swift-cmake-options=-DSWIFT_STDLIB_ENABLE_SIL_OWNERSHIP=FALSE -DCMAKE_EXPORT_COMPILE_COMMANDS=TRUE -DSWIFT_FORCE_OPTIMIZED_TYPECHECKER=FALSE' --lldb-cmake-options= --llbuild-cmake-options= --libdispatch-cmake-options= --foundation-cmake-options= --xctest-cmake-options= --swiftpm-cmake-options= --build-stdlib-deployment-targets all --ninja-bin=/usr/bin/ninja --skip-build-libicu --skip-build-playgroundsupport --build-swift-dynamic-stdlib --build-swift-static-stdlib --build-swift-dynamic-sdk-overlay --build-swift-static-sdk-overlay --skip-build-ios-device --skip-build-ios-simulator --skip-build-tvos-device --skip-build-tvos-simulator --skip-build-watchos-device --skip-build-watchos-simulator --skip-build-android --skip-test-swift --skip-test-cmark --skip-test-lldb --skip-test-llbuild --skip-test-swiftpm --skip-test-xctest --skip-test-foundation --skip-test-libdispatch --skip-test-libicu --skip-test-playgroundsupport --skip-test-linux --skip-test-freebsd --skip-test-cygwin --skip-test-osx --skip-test-ios-host --skip-test-ios-simulator --skip-test-tvos-host --skip-test-tvos-simulator --skip-test-watchos-host --skip-test-watchos-simulator --skip-test-android-host --skip-test-benchmarks --skip-test-optimized --skip-test-optimize-for-size --android-deploy-device-path /data/local/tmp --install-libdispatch --install-foundation --install-swift --install-lldb --install-llbuild --install-xctest --install-swiftpm '--swift-install-components=autolink-driver;compiler;clang-builtin-headers;stdlib;swift-remote-mirror;sdk-overlay;dev' --install-destdir=/home/pi/swift/install --installable-package=/home/pi/swift/swift-4.1.2.tgz --llvm-lit-args=-sv
Building the standard library for: swift-test-stdlib-linux-armv7
cmark: using gold linker
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/cmark-linux-armv7 -- -j4 all
ninja: no work to do.
llvm: using gold linker
symlinking the system headers (/usr/include/c++) into the local clang build directory (/home/pi/swift/build/buildbot_linux/llvm-linux-armv7/include).
+ ln -s -f /usr/include/c++ /home/pi/swift/build/buildbot_linux/llvm-linux-armv7/include
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/llvm-linux-armv7 -- -j4 all
ninja: no work to do.
swift: using gold linker
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/swift-linux-armv7 -- -j4 all swift-test-stdlib-linux-armv7
[1/8] Performing build step for 'libdispatch'
ninja: no work to do.
[2/2] Building HTML documentation
Running Sphinx v1.4.9
loading pickled environment... done
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 0 changed, 0 removed
looking for now-outdated files... none found
no targets are out of date.
build succeeded.
lldb: using gold linker
+ /usr/bin/cmake --build /home/pi/swift/build/buildbot_linux/lldb-linux-armv7 -- -j4 all
[900/900] Linking CXX shared library lib/liblldb-intel-mpxtable.so
llbuild: using gold linker
+ mkdir -p /home/pi/swift/build/buildbot_linux/llbuild-linux-armv7
+ pushd /home/pi/swift/build/buildbot_linux/llbuild-linux-armv7
~/swift/build/buildbot_linux/llbuild-linux-armv7 ~/swift
+ env /usr/bin/cmake -G Ninja -DCMAKE_C_COMPILER:PATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:PATH=/usr/bin/clang++ -DLLVM_VERSION_MAJOR:STRING=5 -DLLVM_VERSION_MINOR:STRING=0 -DLLVM_VERSION_PATCH:STRING=0 -DCLANG_VERSION_MAJOR:STRING=5 -DCLANG_VERSION_MINOR:STRING=0 -DCLANG_VERSION_PATCH:STRING=0 -DCMAKE_MAKE_PROGRAM=/usr/bin/ninja -DCMAKE_EXE_LINKER_FLAGS:STRING=-fuse-ld=gold -DCMAKE_SHARED_LINKER_FLAGS:STRING=-fuse-ld=gold -DC [... cut ...]
Package libcurl was not found in the pkg-config search path.
Perhaps you should add the directory containing `libcurl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libcurl' found
pkg-config error for package libcurl: pkg-config exited with error code 1
./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting
```
I can only spot one `libcurl` in `/home/pi/swift/swift-corelibs-foundation/Foundation/URLSession/libcurl`. Should I add this one to `PKG_CONFIG_PATH`?

Many thanks!

uraimo, 2018-06-25:
Again linuxbrew (the pkg-config of linuxbrew), you could try installing the curl package, since they don't have a separate "libcurl" package, with `brew install curl`. Not really sure if it would work.

Or, following [their instructions](https://github.com/Linuxbrew/legacy-linuxbrew/issues/49), this is what I would do to solve all the issues related to linuxbrew:

        ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Linuxbrew/install/master/uninstall)"
        rm -rf ~/.linuxbrew

:)

Hackberger, 2018-06-26:
After 24 h another error came up.

I removed `brew`.
I ran `clean.sh`.
I ran `build.sh`.

Console output starting at some lines above the error.

```
xctest-build: /home/pi/swift/build/buildbot_linux/swift-linux-armv7/bin/swiftc -emit-library /home/pi/swift/build/buildbot_linux/xctest-linux-armv7/XCTest.o -L /home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/src/.libs -L /home/pi/swift/build/buildbot_linux/foundation-linux-armv7/Foundation -lswiftGlibc -lswiftCore -lFoundation -lm -Xlinker -rpath=\$ORIGIN -o /home/pi/swift/build/buildbot_linux/xctest-linux-armv7/libXCTest.so
xctest-build: mkdir -p /home/pi/swift/build/buildbot_linux/xctest-linux-armv7/static
xctest-build: ar rcs /home/pi/swift/build/buildbot_linux/xctest-linux-armv7/static/libXCTest.a /home/pi/swift/build/buildbot_linux/xctest-linux-armv7/XCTest.o
xctest-build: Done.
swiftpm: using gold linker
+ /home/pi/swift/swiftpm/Utilities/bootstrap --release --swiftc=/home/pi/swift/build/buildbot_linux/swift-linux-armv7/bin/swiftc --sbt=/home/pi/swift/build/buildbot_linux/llbuild-linux-armv7/bin/swift-build-tool --build=/home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7 --foundation=/home/pi/swift/build/buildbot_linux/foundation-linux-armv7/Foundation --libdispatch-source-dir=/home/pi/swift/swift-corelibs-libdispatch --libdispatch-build-dir=/home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7 --xctest=/home/pi/swift/build/buildbot_linux/xctest-linux-armv7
--- bootstrap: note: building stage1
Compile libc.c
Compile Swift Module 'PackageDescription' (6 sources)
Compile Swift Module 'PackageDescription4' (9 sources)
Link clibc
Compile Swift Module 'libc' (1 sources)
Link libc
Compile Swift Module 'POSIX' (11 sources)
Link POSIX
Compile Swift Module 'Basic' (37 sources)
Link PackageDescription
Link PackageDescription4
/home/pi/swift/swiftpm/Sources/Basic/FileSystem.swift:232:16: warning: 'readdir_r' is deprecated
            if readdir_r(dir, &entry, &entryPtr) < 0 {
               ^
Link Basic
Compile Swift Module 'Utility' (19 sources)
Link Utility
Compile Swift Module 'TestSupportExecutable' (1 sources)
Compile Swift Module 'SourceControl' (4 sources)
Compile Swift Module 'PackageModel' (7 sources)
Link /home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/.bootstrap/bin/TestSupportExecutable
Link PackageModel
Compile Swift Module 'PackageLoading' (9 sources)
Link SourceControl
Link PackageLoading
Compile Swift Module 'PackageGraph' (6 sources)
Link PackageGraph
Compile Swift Module 'Build' (7 sources)
Compile Swift Module 'Xcodeproj' (7 sources)
Link Build
Link Xcodeproj
Compile Swift Module 'Workspace' (7 sources)
Link Workspace
Compile Swift Module 'Commands' (14 sources)
Link Commands
Compile Swift Module 'swift_package' (1 sources)
Compile Swift Module 'swift_test' (1 sources)
Compile Swift Module 'swift_build' (1 sources)
Compile Swift Module 'swift_run' (1 sources)
Link /home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/.bootstrap/bin/swift-test
Link /home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/.bootstrap/bin/swift-build
Link /home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/.bootstrap/bin/swift-package
Link /home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/.bootstrap/bin/swift-run
--- bootstrap: note: building runtime v3 target: PackageDescription: 
Compile Swift Module 'PackageDescription' (6 sources)
Link PackageDescription
--- bootstrap: note: building runtime v4 target: PackageDescription4: 
Compile Swift Module 'PackageDescription' (9 sources)
Link PackageDescription
--- bootstrap: note: building self-hosted 'swift-build': env SWIFT_EXEC=/home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/x86_64-unknown-linux/release/swiftc SWIFT_BUILD_PATH=/home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7 /home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/x86_64-unknown-linux/release/swift-build-stage1 --disable-sandbox -Xlinker -rpath -Xlinker $ORIGIN/../lib/swift/linux -Xlinker -L -Xlinker /home/pi/swift/build/buildbot_linux/foundation-linux-armv7/Foundation -Xlinker -rpath -Xlinker /home/pi/swift/build/buildbot_linux/foundation-linux-armv7/Foundation -Xswiftc -I/home/pi/swift/build/buildbot_linux/foundation-linux-armv7/Foundation -Xswiftc -I/home/pi/swift/build/buildbot_linux/foundation-linux-armv7/Foundation/usr/lib/swift -Xlinker -L -Xlinker /home/pi/swift/build/buildbot_linux/xctest-linux-armv7 -Xlinker -rpath -Xlinker /home/pi/swift/build/buildbot_linux/xctest-linux-armv7 -Xswiftc -I/home/pi/swift/build/buildbot_linux/xctest-linux-armv7 -Xlinker -L/home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/src/.libs -Xswiftc -I/home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/src -Xswiftc -I/home/pi/swift/build/buildbot_linux/libdispatch-linux-armv7/src/swift -Xswiftc -I/home/pi/swift/swift-corelibs-libdispatch -Xcc -fblocks -Xswiftc -enable-testing --configuration release --build-tests
*** Error in `/home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7/x86_64-unknown-linux/release/swift-build-stage1': corrupted double-linked list: 0x00d35f18 ***
--- bootstrap: error: build failed with exit status -6
./swift/utils/build-script: fatal error: command terminated with a non-zero exit status 1, aborting
```

What happened? Many thanks!

uraimo, 2018-06-28:
Something similar will happen when the patches were not applied using checkoutrelease.sh, which I'm sure you've run when you started the process.
Is this replicable removing `/home/pi/swift/build/buildbot_linux/swiftpm-linux-armv7` and trying again?