uraimo/buildSwiftOnARM #23: failed to launch REPL

abesmon, 2018-08-28:
I'm having some issues using precompiled binaries for Raspbian stretch

Running it on:
```Linux orangepizero 4.14.65-sunxi #68 SMP Tue Aug 21 19:57:06 CEST 2018 armv7l GNU/Linux```

`swift` invocation returns:
```error: failed to stop process at REPL breakpoint```

`swift -v` invocation returns:
```
Swift version 4.1.1 (swift-4.1.1-RELEASE)
Target: armv7-unknown-linux-gnueabihf
/usr/bin/lldb "--repl=-disable-objc-interop -color-diagnostics"
error: failed to launch REPL process: process launch failed: 'A' packet returned an error: -1
```

`sudo swift -v` returns:
```
Swift version 4.1.1 (swift-4.1.1-RELEASE)
Target: armv7-unknown-linux-gnueabihf
/usr/bin/lldb "--repl=-disable-objc-interop -color-diagnostics"
error: failed to stop process at REPL breakpoint
```

However, I managed to compile some swift projects without crashes, but this REPL message worry me. Is it ok to work with it?

uraimo, 2018-08-30:
Yes, the REPL has never worked on ARM (even in 2.x), but that doesn't impact the behaviour of the compiler.

abesmon, 2018-08-30:
Thanks! Will remember that. isn’t it good to notice that thing in Readme? I think it’s not obvious :/

uraimo, 2018-08-30:
Yup! It makes a lot of sense, doing it now.

uraimo, 2018-08-30:
Let's leave this long standing issue open too.

uraimo, 2019-11-18:
Just for my reference, the issue is still there even on the test version of 5.1.2 that includes clang and libicu.

Cebtenzzre, 2020-01-31:
I commented on [SR-5632](https://bugs.swift.org/browse/SR-5632). See if this command logs anything unexpected:
`lldb --repl=-disable-objc-interop --one-line-before-file 'log enable lldb host'`