uraimo/buildSwiftOnARM #26: [arm32] [4.2] assertion 'hasVal' failed linking Swift.o

Kaiede, 2018-09-17:
I've been hitting this one after fixing the other issue. Don't know how much time I'll get to look at it. I'll see what I can do on my end, but mostly wanted to make sure this gets tracked somewhere and people know it is an issue if coming in new.

```swift: /home/pi/buildSwiftOnARM/llvm/include/llvm/ADT/Optional.h:160: T &&llvm::Optional<swift::ProjectionKind>::operator*() &&[T = swift::ProjectionKind]: Assertion 'hasVal' failed.

2. .      While running pass #418236 SILFunctionTransform "RedundantLoadElimination" on SILFunction "@$Ss18ReversedCollectionV8IteratorVyx_Gs0C8ProtocolsSt4next7ElementQzSgyFTWs16_UnmanagedStringVys6UInt16VG_Tgq5".
 for 'next()' at /home/pi/buildSwiftOnARM/swift/stdlib/public/core/Reverse.swift:91:19```

uraimo, 2018-09-17:
That's the same point I've reached after applying #25. I've checked for recent modifications to name mangling/demangling and there shouldn't  be any.

Kaiede, 2018-09-18:
Probably not recent. I started trying to bisect the 4.2 development history, and the earliest snapshot tag (4.2-DEVELOPMENT-SNAPSHOT-2018-04-23-a) also fails to build in the same spot. Impressive that it's been this long without someone stumbling on it.

Kaiede, 2018-09-22:
Cause by commit 7c096c9f264075d1f38e2621f3659e25c0b78233 (
https://github.com/apple/swift/commit/7c096c9f264075d1f38e2621f3659e25c0b78233)

Reversing it appears to fix the issue, but reveals another one that I can confirm existed prior to this commit. #30

Kaiede, 2018-09-26:
That’s the wron commit, that commit fixed the other stdlib build failure. 

This was introduced between Jan 25 and Jan 21, still tracking down.

closed 2019-10-02 by uraimo