uraimo/buildSwiftOnARM #28: [Foundation][4.2] NSNumber(value: 0.25) as? Bool should return nil

Kaiede, 2018-09-22:
SwiftyJSON is exposing a bug in this commit: https://github.com/apple/swift-corelibs-foundation/commit/6735f210ea25a417ffd2027cf19d611dc64ea551

This looks to have been introduced by accident. It doesn’t account for floating point values that aren't exactly equal to 0.0 and 1.0, as it just truncates it. There's no newer commit here. Foundation 4.2 on Darwin only allows you to do this cast if the number is exactly 0.0 or 1.0.

Kaiede, 2018-09-22:
This should be pretty straight-forward to fix. Should include tests at the same time so they don't break it.

uraimo, 2018-09-23:
Yep, that `.boolValue` should  behave like described: "[A 0 value always means false, and any nonzero value is interpreted as true](https://developer.apple.com/documentation/foundation/nsnumber/1410865-boolvalue)".

Kaiede, 2018-09-23:
Not quite, since this is casting, not using boolValue, and the behavior is different. Apple’s 4.2 Foundation returns nil for any float value other than 0.0 and 1.0. Independent of what boolValue does. Which makes sense since you use NSNumber for boxing values, and casting is how you unbox them.

Kaiede, 2018-09-23:
Fix for the Foundation Behavior: 
https://github.com/apple/swift-corelibs-foundation/pull/1703

Swift Bug around Unconditional Bridging of NSNumber which doesn't match Darwin's Conditional Bridging:
https://bugs.swift.org/browse/SR-8819

uraimo, 2019-02-13:
Included in 4.2.2.

closed 2019-02-13 by uraimo