uraimo/buildSwiftOnARM #33: checkoutRelease.sh

AdrianBinDC, 2018-12-16:
I'm running Raspbian on a Raspberry Pi 3 B+. Here's the output from `uname -a`:

> Linux pi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

The prompt was `[n]`, so I selected `n` and I'm in wait mode to see if it compiles.

When I run `checkoutRelease.sh`, terminal barfs this:
```
pi@pi:~/Downloads/buildSwiftOnARM $ ./checkoutRelease.sh ♻️  Resetting the repositories...
* Cleaning ./swift-stress-tester
HEAD is now at 3b0c86f Merge pull request #24 from nathawes/processrunner-read-data-to-end-of-file
* Cleaning ./swift-xcode-playground-support
HEAD is now at 6c112c2 Merge pull request #31 from cwakamo/migrate-to-swift-4.2
* Cleaning ./icu
HEAD is now at b3054b334d ICU-13640 Merging the removal of Unified Cache Asserts (r41145) to maint-61
* Cleaning ./swift-corelibs-foundation
HEAD is now at 101216ee Merge pull request #1803 from broadwaylamb/master
Removing CoreFoundation/Base.subproj/CFBase.h.rej
Removing CoreFoundation/Base.subproj/CFInternal.h.rej
Removing CoreFoundation/Base.subproj/CFRuntime.c.rej
Removing CoreFoundation/Base.subproj/CFRuntime.h.rej
Removing CoreFoundation/String.subproj/CFString.c.rej
Removing CoreFoundation/String.subproj/CFString.h.rej
Removing Foundation/NSCFString.swift.rej
* Cleaning ./swift-corelibs-libdispatch
HEAD is now at afa6cc3 Merge pull request #434 from compnerd/pic
Removing src/CMakeLists.txt.orig
Removing src/CMakeLists.txt.rej
* Cleaning ./cmark
HEAD is now at 32fa496 Merge pull request #8 from nkcsgexi/update-branch
* Cleaning ./swiftpm
HEAD is now at a3d2d241 Emit duplicate target dependency warnings (#1903)
Removing Sources/Build/Triple.swift.orig
Removing Sources/Build/Triple.swift.rej
Removing Utilities/bootstrap.orig
Removing Utilities/bootstrap.rej
* Cleaning ./clang
HEAD is now at ad7fb41b1d Merge remote-tracking branch 'origin/swift-5.0-branch' into stable
* Cleaning ./ninja
HEAD is now at 253e94c v1.8.2
* Cleaning ./lldb
HEAD is now at 9007e37e82 Merge remote-tracking branch 'origin/swift-5.0-branch' into stable
* Cleaning ./swift
HEAD is now at 9c12fd1c41 Merge pull request #21363 from gottesmm/pr-eb302f1196c943be913dfde41bf571754efa96c8
Removing lib/ClangImporter/ImportDecl.cpp.rej
Removing stdlib/public/core/VarArgs.swift.rej
* Cleaning ./swift-corelibs-xctest
HEAD is now at 8a976c8 Merge pull request #240 from omochi/tls-nsmutabledictionary
* Cleaning ./llbuild
HEAD is now at c1a625b Use Clang attributes to import llbuild enumerations more nicely in Swift
* Cleaning ./swift-integration-tests
HEAD is now at 55971ad Merge pull request #45 from gottesmm/pr-afd41c7a3b1bc1d07a57c374e5e3937f34f68ba7
* Cleaning ./llvm
HEAD is now at f63b283c714 Merge remote-tracking branch 'origin/swift-5.0-branch' into stable
* Cleaning ./compiler-rt
HEAD is now at d4667fe98 Merge remote-tracking branch 'origin/swift-5.0-branch' into stable
* Cleaning ./swift-syntax
HEAD is now at aee69f4 Merge pull request #51 from nkcsgexi/remove-atomic
✳️  Switching all the repositories to swift-4.1.3-RELEASE @ ...
usage: update-checkout [-h] [--clone] [--clone-with-ssh] [--skip-history]
                       [--skip-repository DIRECTORY] [--scheme BRANCH-SCHEME]
                       [--reset-to-remote] [--clean] [--config CONFIG]
                       [--github-comment GITHUB-COMMENT] [--dump-hashes]
                       [--dump-hashes-config BRANCH-SCHEME-NAME]
                       [--tag TAG-NAME] [--match-timestamp] [-j N_PROCESSES]
update-checkout: error: argument --tag: expected one argument
✅  Applying the required patches...
* Applying patches to ./swift-corelibs-foundation
patching file CoreFoundation/Base.subproj/CFBase.h
Hunk #1 FAILED at 683.
1 out of 1 hunk FAILED -- saving rejects to file CoreFoundation/Base.subproj/CFBase.h.rej
patching file CoreFoundation/Base.subproj/CFInternal.h
Hunk #1 FAILED at 412.
1 out of 1 hunk FAILED -- saving rejects to file CoreFoundation/Base.subproj/CFInternal.h.rej
patching file CoreFoundation/Base.subproj/CFRuntime.c
Hunk #1 FAILED at 387.
1 out of 1 hunk FAILED -- saving rejects to file CoreFoundation/Base.subproj/CFRuntime.c.rej
patching file CoreFoundation/Base.subproj/CFRuntime.h
Hunk #1 FAILED at 193.
1 out of 1 hunk FAILED -- saving rejects to file CoreFoundation/Base.subproj/CFRuntime.h.rej
patching file CoreFoundation/String.subproj/CFString.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Hunk #1 FAILED at 245.
1 out of 1 hunk FAILED -- saving rejects to file CoreFoundation/String.subproj/CFString.c.rej
patching file CoreFoundation/String.subproj/CFString.h
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file CoreFoundation/String.subproj/CFString.h.rej
patching file Foundation/NSCFString.swift
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file Foundation/NSCFString.swift.rej
patching file Foundation/NSSwiftRuntime.swift
Hunk #1 succeeded at 88 with fuzz 2 (offset 1 line).
* Applying patches to ./swift-corelibs-libdispatch
patching file src/CMakeLists.txt
Hunk #1 succeeded at 252 with fuzz 2 (offset 45 lines).
Hunk #2 FAILED at 218.
1 out of 2 hunks FAILED -- saving rejects to file src/CMakeLists.txt.rej
* Applying patches to ./swiftpm
patching file Sources/Build/Triple.swift
Hunk #1 FAILED at 34.
Hunk #2 FAILED at 106.
2 out of 2 hunks FAILED -- saving rejects to file Sources/Build/Triple.swift.rej
patching file Utilities/bootstrap
Hunk #1 FAILED at 893.
1 out of 1 hunk FAILED -- saving rejects to file Utilities/bootstrap.rej
patching file Sources/Build/Triple.swift
Hunk #1 FAILED at 33.
Hunk #2 FAILED at 109.
2 out of 2 hunks FAILED -- saving rejects to file Sources/Build/Triple.swift.rej
patching file Utilities/bootstrap
Hunk #1 FAILED at 891.
1 out of 1 hunk FAILED -- saving rejects to file Utilities/bootstrap.rej
* Applying patches to ./swift
patching file lib/ClangImporter/ImportDecl.cpp
Hunk #1 FAILED at 273.
1 out of 1 hunk FAILED -- saving rejects to file lib/ClangImporter/ImportDecl.cpp.rej
patching file stdlib/public/core/CTypes.swift
Hunk #1 succeeded at 218 with fuzz 2 (offset 14 lines).
Hunk #2 succeeded at 261 with fuzz 2 (offset 15 lines).
patching file stdlib/public/core/VarArgs.swift
Hunk #1 FAILED at 85.
Hunk #2 succeeded at 645 with fuzz 1 (offset 178 lines).
1 out of 2 hunks FAILED -- saving rejects to file stdlib/public/core/VarArgs.swift.rej
patching file stdlib/public/SwiftShims/RefCount.h
Hunk #1 succeeded at 1133 (offset -169 lines).
* Applying patches to ./llvm
patching file lib/ExecutionEngine/RuntimeDyld/RuntimeDyldELF.cpp
Hunk #1 succeeded at 525 with fuzz 2 (offset 45 lines).
Hunk #2 succeeded at 1246 (offset -81 lines).
pi@pi:~/Downloads/buildSwiftOnARM $ 
```

AdrianBinDC, 2018-12-17:
...and...KABOOM after 3 hours...

```
[2108/3019] Building CXX object tools/...keFiles/clangSema.dir/ParsedAttr.cpp.o
In file included from /home/pi/Downloads/buildSwiftOnARM/llvm/tools/clang/lib/Sema/ParsedAttr.cpp:169:
tools/clang/include/clang/Sema/AttrParsedAttrImpl.inc:1081:19: warning: unused variable 'S' [-Wunused-variable]
  if (const auto *S = dyn_cast<FunctionDecl>(D))

```

Are there any installs for Raspbian out there that work under the assumption the user eats paste and crayons?

futurejones, 2018-12-17:
There is a link in the readme to a pre-built Raspbian binary - https://www.dropbox.com/s/e257cvg23ghe2dt/swift-4.1.3-RPi23-RaspbianStretch.tgz?dl=0
or if you want an `apt-get install` there is 4.1.2 available here - https://packagecloud.io/swift-arm/debian/packages/raspbian/stretch/swift-4.1-RPi23-RaspbianStretch_4.1.2_armhf.deb
Make sure you follow the instructions to installed the repository before you try to `sudo apt-get install swift4`
I will be updated PackageCloud to 4.1.3 soon.

uraimo, 2018-12-17:
@AdrianBinDC my fault, I left out the TAG variable when I updated the checkout script, it should work now (but it would require a complete rebuild since without this you were building from the master branch w/o patches).

closed 2018-12-17 by uraimo

AdrianBinDC, 2018-12-18:
@uraimo Thank you!

AdrianBinDC, 2018-12-18:
@futurejones Thank you. I'm going to give that a whirl, too. Thank you.

AdrianBinDC, 2018-12-18:
> @AdrianBinDC There is a link in the readme to a pre-built Raspbian binary - https://www.dropbox.com/s/e257cvg23ghe2dt/swift-4.1.3-RPi23-RaspbianStretch.tgz?dl=0

Thank you. I dl'ed the file onto my machine and untarred it...where do I put the `usr` directory (crayons and paste, remember? :) )