uraimo/buildSwiftOnARM #48: [armv6][5.0] Prebuilt binary runs, fails to compile with unknown error

idrougge, 2019-05-25:
I downloaded and extracted the ARMv6 5.0 binary according to instructions to my RPi 1b with Raspbian, installing the additional required packages with `sudo apt install clang-3.8 libicu-dev libcurl4-nss-dev` and added the extracted `usr` directory to the path.

Running `swift test.swift` will run the source file in question, but compiling to a file with `swiftc test.swift` always fails with `<unknown>:0: error: unable to execute command: <unknown>` at the end of compilation.

uraimo, 2019-05-25:
This should 99% fix it, I suppose you didn't register clang-3.8 with the alternatives for `clang`:

```
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/bin/clang++-3.8 100
sudo update-alternatives --install /usr/bin/clang clang /usr/bin/clang-3.8 100
```

idrougge, 2019-05-25:
This works perfectly. Should this be added to the readme?

uraimo, 2019-05-25:
I thought it was already there but it wasn't... Just added :)

closed 2019-05-25 by idrougge