uraimo/buildSwiftOnARM #51: [aarch64] at 5.0.1 or still at 4.2.3?

marc-medley, 2019-06-26:
Is `aarch64` at 5.0.1 or still at 4.2.3?

It would be helpful if `aarch64` guidance (and latest applicable link) was still on the README.md **Prebuilt binaries** table.

[image]

uraimo, 2019-06-26:
It's at 5.0.1, it doesn't actually need patches (other than [this](https://github.com/uraimo/buildSwiftOnARM/tree/master/swift.diffs/aarch64) on 18.04) and you can get it from [here](https://github.com/futurejones/swift-arm64), I just stopped building it because it's a bit cumbersome from the logistic point of view.

Right now I'm streamlining the process to get new builds faster and I don't exclude I'll be able to bring back aarch64.

marc-medley, 2019-06-26:
Thanks for the quick reply.

> I don't exclude I'll be able to bring back aarch64.

The wording of the above sentence confuses me a bit...

Would a correct understanding be to _not expect_ `aarch64` in future [uraimo/buildSwiftOnARM](https://github.com/uraimo/buildSwiftOnARM) releases?

And that, going forward, one should plan on using an alternate source for `aarch64` Swift releases such as [futurejones/swift-arm64](https://github.com/futurejones/swift-arm64)?

> (other than [this](https://github.com/uraimo/buildSwiftOnARM/tree/master/swift.diffs/aarch64) on 18.04)

The "this" link <https://github.com/uraimo/buildSwiftOnARM/tree/master/swift.diffs/aarch64> lands on a 404 not found page.

uraimo, 2019-06-26:
> is planned to be included in future uraimo/buildSwiftOnARM releases?

aarch64 will always be supported by the toolchain, so if you'll need additional patches to compile Swift on that platform you'll find them here. IRT the binaries, those take time to be built considering the increasing number of variations we have, so going forward I'll have to choose which combinations to keep and the rest will be compiled by the users using the scripts.
But yes, aarch64 binaries for the latest Ubuntu LTS release should be available in the future, probably even starting with 5.0.1.

But note that they will always be provided in the current plain tgz format, if you need something distributed more properly via apt and with automatic dependencies management you should consider swift-arm64. I will also never provide Swift snapshots before an official release has been tagged, something that futurejones usually does in addition to provide binaries for older LTSes like 16.04. 

> lands on a 404 not found page.

Sorry, I've updated the script a few hours ago :)
That link is now: https://github.com/uraimo/buildSwiftOnARM/tree/master/swift.diffs/aarch64ubuntu18.04

uraimo, 2019-07-09:
Link to the arm64 version added, but considering how unstable my build board is with Ubuntu 64 (1 week of crashes), 5.0.1 could be the last version I provide binaries for.

closed 2019-07-09 by uraimo