uraimo/buildSwiftOnARM #52: PATH not persisting

AdrianBinDC, 2019-06-30:
I downloaded the precompiled 5.0.1 version of Swift for Raspbian Buster. Two questions which I know are super basic 101 questions.

1) Where on the disk should I put the unpacked `usr`directory containing the Swift files?
2) What do I need to do to get the PATH to persist? I tried putting the lines in referenced on the front page of this repo, but that doesn't work for me. I rest assured it's completely my fault.

I was able to get it to work by typing export `PATH=$PATH:/home/pi/Swift/usr/bin`, however that doesn't persist between launches of terminal.

AdrianBinDC, 2019-06-30:
Disregard. I rebooted the Raspberry Pi instead of the terminal and it worked.

closed 2019-06-30 by AdrianBinDC

uraimo, 2019-06-30:
Btw, even if the readme right now recommend something different, I'd suggest to put it in `/opt/` renaming `usr` in `swift-5.0.1` and then create a symbolic link to it with `ln -s /opt/swift-5.0.1 /opt/swift`.
And add `/opt/swift/bin` to the path to instead of keeping it in the home directory.
This way you can keep different releases if you need it without messing up your home directory.