uraimo/buildSwiftOnARM #56: [5.1] SPM: error: toolchain is invalid: could not find the `swiftc` at expected path

uraimo, 2019-10-07:
Noticed this on a container with armv7 Ubuntu 18.04, when running `swift test` and likely other spm commands:

```
error: toolchain is invalid: could not find the `swiftc` at expected path /github/workspace/usr/bin/swiftc
```

No issues on Debian Buster.
Needs investigation, could be just a fluke.

uraimo, 2019-10-07:
Like #57 I haven't verified by I suspect it depends on the fact that it was running in a docker container on Github CI.

uraimo, 2019-10-07:
Working fine on a Docker container with `overlay2` storage driver.

uraimo, 2019-10-07:
Closing this since it's an issue specific to GitHub CI or their docker config (still in beta).

closed 2019-10-07 by uraimo

maximkrouk, 2020-01-02:
Have the same issue on VPS without docker, `swift build` causes this error

uraimo, 2020-01-02:
@maximkrouk Hi! can you paste the `uname -a` for that VPS and tell me which vps provider are you using?
Thanks!

maximkrouk, 2020-01-02:
The provider is [Hostinger](hostinger.com), VPS is running Ubuntu Linux 18.04, I was tried to run `swift build -c release --disable-sandbox` on Vapor[vapor.codes] toolbox.
`Linux <hostname> 4.15.0 #1 SMP Thu Jun 27 15:10:55 MSK 2019 x86_64 x86_64 x86_64 GNU/Linux`

maximkrouk, 2020-01-03:
No errors after changing my VPS system to Ubuntu16.04 c:

xnutsive, 2020-08-28:
I've just hit this with Ubuntu 18.04 in Docker on a mac. I'll try the storage driver you suggested, @uraimo, and report back.