uraimo/buildSwiftOnARM #60: The Swift 5.x toolchain on arm32 does not contain clang/libicu

uraimo, 2019-11-14:
The usual preset `buildbot_linux,no_test` does not work for arm32 right now, near the end of the process SPM complains that it can't find libicu (correct, it's not being copied over) when building the bootstrap SPM.

The result of this is that Swift 5.x on armv6/armv7 still requires the clang and libicu dependencies, but depending on an old system version of clang or an obsolete release of libicu (e.g. on distributions derived from Stretch/Buster) leads to issues like [#148](https://github.com/apple/swift-nio-ssl/issues/148).

uraimo, 2019-11-14:
Skipping SPM and specifying a "minimal" set of build options similar to `buildbot_linux,no_test` the toolchain has everything we expect (minus SPM obviously).
Considering that we kept the build command more or less the same in the last few years, we were missing a few options and more importantly `--llvm-install-components=`, that's what actually specifies which llvm sub-components should be installed.

```
./swift/utils/build-script \
    -R \
    --build-subdir buildbot_linux --install-prefix=/usr \
    --install-destdir=${INSTALL_DIR} --installable-package=${PACKAGE} \
    --lldb --llbuild --xctest --libicu --libcxx --foundation --libdispatch \
    --skip-test-cmark --skip-test-lldb --skip-test-swift --skip-test-llbuild --skip-test-swiftpm --skip-test-xctest --skip-test-foundation --skip-test-libdispatch --skip-test-playgroundsupport --skip-test-libicu --skip-test-indexstore-db --skip-test-sourcekit-lsp \
    -- --install-libdispatch --install-foundation --install-swift --install-lldb --install-llbuild --install-xctest --install-libicu --install-libcxx \
    '--swift-install-components=autolink-driver;compiler;clang-builtin-headers;stdlib;swift-remote-mirror;sdk-overlay;dev' \
    '--llvm-install-components=llvm-cov;llvm-profdata;clang;clang-headers;compiler-rt;clangd' \
    --build-swift-static-stdlib --build-swift-static-sdk-overlay
```

uraimo, 2019-12-04:
Note to self: Even the silly workaround of building without SPM using the command above, over an `install` dir already populated with an SPM generated the usual way doesn't work. 
Still get this: 
```
./buildSwiftOnARM/install/usr/bin/swift build
/home/buildSwiftOnARM/install/usr/bin/swift-build: error while loading shared libraries: libicudataswift.so.61: cannot open shared object file: No such file or directory
```
because:
```
$ ldd /home/buildSwiftOnARM/install/usr/bin/swift-build
        linux-vdso.so.1 (0xbee28000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb69a5000)
        ...
        libatomic.so.1 => /usr/lib/arm-linux-gnueabihf/libatomic.so.1 (0xb56cc000)
        libicuuc.so.60 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.60 (0xb5596000)
        libicui18n.so.60 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.60 (0xb53cd000)
        libicuucswift.so.61 => /home/buildSwiftOnARM/install/usr/bin/../lib/swift/linux/libicuucswift.so.61 (0xb5218000)
        libicui18nswift.so.61 => /home/buildSwiftOnARM/install/usr/bin/../lib/swift/linux/libicui18nswift.so.61 (0xb4f6f000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb4f59000)
        libllbuild.so => /home/buildSwiftOnARM/install/usr/bin/../lib/swift/pm/llbuild/libllbuild.so (0xb4df1000)
        libicudata.so.60 => /usr/lib/arm-linux-gnueabihf/libicudata.so.60 (0xb3438000)
        libicudataswift.so.61 => not found
        libicudataswift.so.61 => not found
```
But at list it's there:
```
$ ls /home/buildSwiftOnARM/install/usr/lib/swift/linux/
armv7/                        libicudataswift.so            libicuucswift.so.61
libBlocksRuntime.so           libicudataswift.so.61         libicuucswift.so.61.1
libFoundation.so              libicudataswift.so.61.1       libswiftCore.so
libFoundationNetworking.so    libicui18nswift.so            libswiftDispatch.so
libFoundationXML.so           libicui18nswift.so.61         libswiftGlibc.so
libXCTest.so                  libicui18nswift.so.61.1       libswiftRemoteMirror.so
libdispatch.so                libicuucswift.so              libswiftSwiftOnoneSupport.so
```

Arasthel, 2020-03-21:
I think I got it to build with --libicu.

First, I modified `./icu/icu4c/source/config/mh-linux` as explained here, removing the `LDFLAGSICUDT` variable altogether or assigning nothing to it:

https://www.raspberrypi.org/forums/viewtopic.php?p=392071&sid=6138a1969a19cbc0be9d0d8641502aae#p392071

Then I tried building without the --libicu modifier and I got the `swift-build: error while loading shared libraries: libicudata.so.65: cannot open shared object file: No such file or directory` error. However, I copied `libicudata.so*` to /usr/lib, as it's the prefix provided in `build.sh` and the error was gone and ldd showed libicudata.so.65 properly linked.

I tried again with --libicu, got the same error, copied libicuswift* to /usr/lib and:

```
pi@raspberrypi:~/buildSwiftOnARM $ ldd build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/swift-build
	linux-vdso.so.1 (0xbebe5000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6a3c000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6a12000)
	libutil.so.1 => /lib/arm-linux-gnueabihf/libutil.so.1 (0xb69ff000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb69ec000)
	libswiftCore.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/libswiftCore.so (0xb6448000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb63c6000)
	libswiftGlibc.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/swift-linux-armv7/lib/swift/linux/libswiftGlibc.so (0xb63b9000)
	libFoundation.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/../lib/swift/linux/libFoundation.so (0xb5ae1000)
	libswiftDispatch.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/../lib/swift/linux/libswiftDispatch.so (0xb5ab7000)
	libdispatch.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/../lib/swift/linux/libdispatch.so (0xb5a74000)
	libBlocksRuntime.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/../lib/swift/linux/libBlocksRuntime.so (0xb5a61000)
	libllbuildSwift.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/llbuild-linux-armv7/lib/libllbuildSwift.so (0xb5a43000)
	libsqlite3.so.0 => /usr/lib/arm-linux-gnueabihf/libsqlite3.so.0 (0xb593b000)
	libncurses.so.6 => /lib/arm-linux-gnueabihf/libncurses.so.6 (0xb590d000)
	libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0xb58dc000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb5795000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb5768000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb561a000)
	/lib/ld-linux-armhf.so.3 (0xb6fc2000)
	libatomic.so.1 => /usr/lib/arm-linux-gnueabihf/libatomic.so.1 (0xb5601000)
	libicuuc.so.65 => /usr/local/lib/libicuuc.so.65 (0xb5432000)
	libicui18n.so.65 => /usr/local/lib/libicui18n.so.65 (0xb516a000)
	libicuucswift.so.65 => /usr/lib/libicuucswift.so.65 (0xb4f95000)
	libicui18nswift.so.65 => /usr/lib/libicui18nswift.so.65 (0xb4cbd000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb4ca4000)
	libllbuild.so => /home/pi/buildSwiftOnARM/build/buildbot_linux/llbuild-linux-armv7/lib/libllbuild.so (0xb4b38000)
	libicudata.so.65 => /usr/lib/libicudata.so.65 (0xb3078000)
	libicudataswift.so.65 => /usr/lib/libicudataswift.so.65 (0xb15b8000)
```

This however did not solve the issues with Strings manipulation being really time expensive on 32bit as I hope it would (issue #63), so I guess it's a different issue altogether.

Kaiede, 2020-03-26:
If the library needs to be moved to /usr/lib to work, then something related to the RUNPATH is likely not correct somewhere. Because really all that moving files there does is put it in the LDPATH, but Swift is designed to rely on RUNPATH to make a self-contained tarball.

I'd be very curious what you get from `chrpath build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/swift-build` (may need to install the tool first), and how that compares to where `libicudataswift.so.65` and `libicudata.so.65` wind up in the `build/` folder.

Because right now it looks like it only searches in:
* `.`
* `../lib/swift/linux`
* `../lib/swift/pm/llbuild`

So if libicudata is under some other path instead, it should be clearer what's going on at that point.

Arasthel, 2020-03-28:
The chrpath command returns:

```
pi@raspberrypi:~/buildSwiftOnARM $ chrpath build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/swift-build
build/buildbot_linux/swiftpm-linux-armv7/armv7-unknown-linux-gnueabihf/release/swift-build: RUNPATH=/home/pi/buildSwiftOnARM/build/buildbot_linux/swift-linux-armv7/lib/swift/linux:/usr/lib/swift:$ORIGIN:$ORIGIN/../lib/swift/linux:/home/pi/buildSwiftOnARM/build/buildbot_linux/swiftpm-linux-armv7/.bootstrap/lib/swift/linux:$ORIGIN/../lib/swift/pm/llbuild:/home/pi/buildSwiftOnARM/build/buildbot_linux/llbuild-linux-armv7/lib
```