uraimo/buildSwiftOnARM #61: Could not construct an expression context for the REPL (Raspberry Pi 4, Debian Buster)

JetForMe, 2019-11-17:
I'm having trouble figuring out what's going wrong with this. I upgraded my system today. I haven't worked with this in months, but I think it was working with a 5.0.2 build of your tools. I was having a similar problem with that version, so I updated to 5.1.1, and the error I get now is:

```
rmann@rpi4:~ $ swift --version
Swift version 5.1.1 (swift-5.1.1-RELEASE)
Target: armv7-unknown-linux-gnueabihf
rmann@rpi4:~ $ swift
error: Could not construct an expression context for the REPL.

rmann@rpi4:~ $ 
```

With 5.0.2 I got:

```
rmann@rpi4:~/01_hello_SDL $ swift
error: failed to stop process at REPL breakpoint
```

I'm not sure what's going on.

uraimo, 2019-11-18:
Hi, thanks for opening this so I can post the current status somewhere.
Yes, it’s a known issue we’ve had since the beginning(2.x) see #23 , the REPL doesn’t work but everything else do(swiftc, SPM,...).

With the last few 5.1.x the error changed and is now aligned with the output we got for the old [SR-5632](https://bugs.swift.org/browse/SR-5632).

JetForMe, 2019-11-18:
Ah, I guess I missed it in the Readme where it says the REPL doesn't work. Pity.

uraimo, 2019-11-18:
Closing this as a duplicate of #23.

closed 2019-11-18 by uraimo