uraimo/buildSwiftOnARM #62: swiftc IF issue

ScriptBasic, 2019-11-27:
There seems to be a problem with an IF compare with **swiftc**. The **swift** interpreter works fine.

// Swift - 1mil.swift

var accum = 0
var milvar = 1000000

for _ in 1...1000000 {
  accum += 1
  milvar -= 1
  if accum == milvar {
    print("Middle")
  }  
}

print("Plus:  ", accum)
print("Minus: ", milvar)

**Interpretive**

pi@RPi4B:~/swift-dev/examples $ /usr/bin/time swift 1mil.swift
Middle
Plus:   1000000
Minus:  0
0.80user 0.05system 0:00.86elapsed 99%CPU (0avgtext+0avgdata 78392maxresident)k
240inputs+0outputs (2major+4179minor)pagefaults 0swaps
pi@RPi4B:~/swift-dev/examples $
[/tt]

**Compiled**

pi@RPi4B:~/swift-dev/examples $ swiftc 1mil.swift
pi@RPi4B:~/swift-dev/examples $ ls -l 1mil
-rwxr-xr-x 1 pi pi 14524 Nov 25 17:08 1mil
pi@RPi4B:~/swift-dev/examples $ /usr/bin/time ./1mil
Plus:   1000000
Minus:  0
0.24user 0.01system 0:00.25elapsed 99%CPU (0avgtext+0avgdata 6428maxresident)k
0inputs+0outputs (0major+268minor)pagefaults 0swaps
pi@RPi4B:~/swift-dev/examples $

uraimo, 2019-11-28:
I can't replicate this issue, which Swift release are you using? My prebuilt 5.1.1?

ScriptBasic, 2019-11-29:
pi@RPi4B:~ $ swiftc -v
Swift version 5.1.1 (swift-5.1.1-RELEASE)
Target: armv7-unknown-linux-gnueabihf
pi@RPi4B:~ $ 

I'm running on a RPi 4B 5GB with 32 bit Raspbain Buster.

ScriptBasic, 2019-11-30:
I'm building the 5.12 release from source. I had to move the **llvm-project** contents down to the base directory for **build.sh** to run. Since I'm running on a RPi 4B with $GB, I passed on the swap file section of instructions. 

I'll let you know how it turns out. (still building)

ScriptBasic, 2019-11-30:
The 5.12 build corrected the IF compare issue. Would you like a  link to the **swift-5.1.2_armv7.tgz** file on my server?

uraimo, 2019-11-30:
Replies in reverse order ;)

> The 5.12 build corrected the IF compare issue. 

Weird, just to keep this logged somewhere, if you still have the 5.1.1 `1mil` binary with the issue and if it's not too big, can you upload it here?
For the toolchain, I already have a few builds ready but I'm holding off because I'd like to find a workaround for #60 .

>  I had to move the llvm-project contents down to the base directory for build.sh to run. 

Starting from scratch works too, good to know that a manual copy works too, thanks.

> I'm running on a RPi 4B 4GB with 32 bit Raspbain Buster.

This was the same configuration I tried it on, very strange issue.

ScriptBasic, 2019-12-01:
Your request was too late. I no longer have 5.1.1 installed or the bad **1mil** that was generated.

uraimo, 2019-12-01:
Oh ok, no problem.

closed 2019-12-01 by uraimo