uraimo/buildSwiftOnARM #67: File not found

NSCabezon, 2021-01-11:
Hi, I've installed precompiled swift binaries and tried to run a simple program I already had that prints a hello world and lists files from current path ("ls") and I get the following output:

```
Hello, world!
Fatal error: Error Domain=NSCocoaErrorDomain Code=260 "The file doesn’t exist.": file /home/buildSwiftOnARM/swift-corelibs-foundation/Foundation/Process.swift, line 331
Instrucción ilegal
```

any idea of why this is happening?

regards,
Iván

uraimo, 2021-01-11:
Hi! Can you share the example that you are trying to run?

NSCabezon, 2021-01-11:
I believe the 

```
print("Hello, world!")

let task = Process()
task.launchPath = "ls"
task.arguments = ["-la"]

let pipe = Pipe()
task.standardOutput = pipe
task.launch()
```

when running it says: 

```
/home/pi/PiSwift/Sources/HelloWorld/main.swift:6:6: warning: 'launchPath' is deprecated: renamed to 'executableURL'
task.launchPath = "ls"
     ^
/home/pi/PiSwift/Sources/HelloWorld/main.swift:6:6: note: use 'executableURL' instead
task.launchPath = "ls"
     ^~~~~~~~~~
     executableURL
/home/pi/PiSwift/Sources/HelloWorld/main.swift:11:6: warning: 'launch()' is deprecated: renamed to 'run'
task.launch()
     ^
/home/pi/PiSwift/Sources/HelloWorld/main.swift:11:6: note: use 'run' instead
task.launch()
     ^~~~~~
     run
[4/4] Linking HelloWorld
```

thanks!

uraimo, 2021-01-11:
`ls` is an actual binary, not something bash provides, so modify your example with `/bin/ls`.
If you try to run it with "ls" the Process will throw, leading to the error you seen, that "Illegal Instruction" is the result of the attempt to dump the stack... that didn't work.

Updated example:
```
import Foundation

print("Hello, world!")

let task = Process()
task.executableURL = URL(fileURLWithPath: "/bin/ls")
task.arguments = ["-la"]

let pipe = Pipe()
task.standardOutput = pipe
task.standardError = pipe
try task.run()

print(String(decoding: pipe.fileHandleForReading.readDataToEndOfFile(), as: UTF8.self) ) 
```

closed 2021-01-18 by uraimo