uraimo/buildSwiftOnARM #68: [5.4] Not implemented relocation type on swift run

uraimo, 2021-05-31:
Spotted with a test armv6 build, need to verify if it's there for the classic armv7 build too.

```
pi@raspberrypi:~ $ ./usr/bin/swift run test.swift
warning: 'swift run file.swift' command to interpret swift files is deprecated; use 'swift file.swift' instead
Not implemented relocation type!
UNREACHABLE executed at /home/buildSwiftOnARM/llvm-project/llvm/lib/ExecutionEngine/RuntimeDyld/RuntimeDyldELF.cpp:512!
Please submit a bug report (https://swift.org/contributing/#reporting-bugs) and include the project and the crash backtrace.
Stack dump:
0.	Program arguments: /home/pi/usr/bin/swift-frontend -frontend -interpret test.swift -disable-objc-interop -color-diagnostics -module-name test
1.	Swift version 5.4-dev (LLVM 8ee67e3af469eb3, Swift fb774de49c67eed)
2.	While running user code "test.swift"
Aborted
```

mickeyl, 2021-06-30:
Did you solve this issue yet? I've noticed you updated the readme, but the links to the archives still point to _nirvana_. :-)

uraimo, 2021-07-23:
Not yet, but it could just be the result of an armv6 patch not being applied ,see #69 for the links (need time).